iNavFlight/OpenTX-Telemetry-Widget #5: BECAUSE MY LUA INAV DOES NOT WORK, GIVE ME RECOMMEND

elyfranco, 2021-08-07:
Dear modeling friends, I ask you to help me get my lua inav to work on my taranis X9D 2019SE transmitter. As described, I downloaded the unzipped LuaTelemetry-Taranis-en.zip file, then I inserted the three files in the telemetry directory of the SD, then as recommended I did the new search for the sensors and reading its value next to each sensor. Then I ran the lua ivav and all the sensors and artificial horizon appeared on the transmitter display. pity that I do not read any sensor value marks everything to zero and the horizon does not move. I ask you the courtesy to tell me what mistake I might have made and why the installed lua does not work. Thanks so much for your time

nm17, 2021-08-07:
@elyfranco can you provide me more info or a way for me to message you privately, so we can figure out what is wrong faster?

elyfranco, 2021-08-07:
Thank you for your interest and sorry for the late reply. Despite my bad English, however, I have carefully followed how to install the files to have the correct functioning of the lua. My Transmitter is a Taranis X9D 2019 SE as inav fireware is 2.6.1 and open TX is 2.3.13 and F.C.  405 wing. From what I understand for the installation it is necessary to download the LUATELEMETRY file, decompress it and copy the two Scripts and Widgets files to the SD card directory. these are the operations I performed. Thanks Franco

nm17, 2021-08-09:
@elyfranco does it work now?

elyfranco, 2021-08-09:
Hi no it works, I don't understand why in F.C. The Matek 405 wing does not work and in F.C. 722 Wing works, know that I use the same Taranis X9D 2019 SE transmitter, the same Open TX and the same Inav 2.6.1 fireware.
Since you are so helpful, I ask you, I tried to install Yaapu lua but I am not clear about the instructions. Would you be available to show me step by step how to install and where to insert the various files. Thanks for your time, good day Franco.

nm17, 2021-08-09:
Thanks for clearing that up, you can contact me privately on my email `me@nm17.dev` if you want some help with Yaapu.

I'll close this issue as it seems that you figured everything out on your own.

closed 2021-08-09 by nm17

elyfranco, 2022-02-04:
Hello greetings everyone, I ask for your help because I have little experience but I wanted to install telemetry on my Taranis x9D 2019 SE. On my FPV flight plane I have installed a Matek 722 flight controller with Crossfire firmware 4.0 receiver. I downloaded the latest version of LuaTelemetry_v2.0.1.zip telemetry from github.com. subsequently opened the file inside there are two folders WIDGET -SCRIPTS that I have inserted in the directory of the sim. Subsequently inserted the sim in the transmitter carried out a search for new sensors on the transmitter screen I see the graphics but I do not read any value of the recognized sensors, a message comes out that says not ok. Where I may have made mistakes, someone is willing to help me make sure everything runs smoothly. Thanks for your time, sorry if my english is not perfect.