iNavFlight/OpenTX-Telemetry-Widget #7: Possible issue with colors in EdgeTX

jfrickmann, 2021-09-10:
We have made some changes to the way that colors are handled in EdgeTX, and we have become aware that this causes an issue, if hard coded RGB565 color values are used instead of values returned by `lcd.RGB(r,g,b)`.
The solution is to replace e.g.
`color1 = 0x0AB1` by `color1 = lcd.RGB(8,84,136)`.
I apologize for the inconvenience, and hope that you understand!

nm17, 2021-09-14:
Can you link the commit that caused this problem? @jfrickmann

jfrickmann, 2021-09-14:
It is not a commit or change in iNav. We made a change to colors in EdgeTX that caused a problem for some others, and I am just giving you a heads up that this may also be a problem for iNav. I do not use it myself, and since you do not publish the source code, I cannot verify for my self whether it is a problem. But the binary format of RGB color codes has been changed, and therefore, hard coded RGB color constants no longer work.

nm17, 2021-09-16:
> I do not use it myself, and since you do not publish the source code, I cannot verify for my self whether it is a problem.

@jfrickmann INav and this project is opensource.

shellixyz, 2021-09-16:
Thanks for the heads up. We were asking what the change in EdgeTX was, not in iNav. Not sure why you're saying the source code is not available since you just posted in the repository hosting it.

jfrickmann, 2021-09-18:
Sorry, I was looking somewhere and found a .luac file, so I assumed that you only published compiled files like some groups do. I looked again, and could see that you do have the source code here - my bad!
I can also see that you sometimes use a literal value with setColor, and this will not work in EdgeTX because we have changed the binary format of colors, to make it possible to use RGB colors and theme colors interchangeably.
I have written up a description here: https://luadoc.edgetx.org/part_ii_-_opentx_lua_api_programming_guide/drawing-flags-and-colors

stronnag, 2021-09-19:
To confirm, this is a problem for the inav telemetry widget; it also corrupts the main ETX screen when you move off the inav widget.

wx4cb, 2021-09-19:
to be honest, in some ways, i'm inclined to say this is not a lua issue but an edge tx issue... but i'm also inclined to say that you shouldnt really be hardcoding anything without an easy way to change it (eg constants in a header, or API calls for colours etc)

stronnag, 2021-09-21:
> to be honest, in some ways, i'm inclined to say this is not a lua issue but an edge tx issue..

It's an EdgeTX feature. The ETX API is so more powerful and elegant than the OTX API; and much easier to use in a graphically complex widget like this.

Anyway, for now there's a **EdgeTX only** [fork](https://github.com/stronnag/EdgeTX-INAV-Telemetry-Widget) which appears to work correctly with ETX 2.5rc1 and RM TX16S (horus/T16 et al) at least.

stronnag, 2021-09-27:
Fixed by #10

closed 2021-09-27 by nm17