iNavFlight/OpenTX-Telemetry-Widget #21: How to enter the menu?

kasatka60, 2021-11-13:
I have a jumper t12 and t-lite (OpenTX 2.3.14). How do I get to the settings menu?

stronnag, 2021-11-13:
Does [this](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/17) help

kasatka60, 2021-11-13:
I pressed enter, then return. Does not work.

stronnag, 2021-11-13:
I'll have a look via the simulator (don't have the radio).

kasatka60, 2021-11-13:
Maybe I am pushing the wrong way? Doesn't work in the simulator.

stronnag, 2021-11-13:
> Maybe I am pushing the wrong way? Doesn't work in the simulator.

It won't work in the simulator until we fix it!

wx4cb, 2021-11-13:
If it is the same as the normal lua, then hold the throttle to the middle
and yaw right until the screen comes up

kasatka60, 2021-11-13:
Does not work.

stronnag, 2021-11-13:
The stick commands will only work on the large colour screen radios (Horus and friends (TX16S et al) and NV14.

wx4cb, 2021-11-13:
sorry, my bad. It’s top right with the left stick

julianschweizer, 2021-12-15:
This still does not work for my Jumper T-Lite, since "EVT_ROT_LONG" only works for radios with rotary encoder. Jumper T-Lite needs "or event == EVT_DOWN_LONG" (Which is the "Down" key) to enter menu.

Is there a way to  use a virtual key like "EVT_VIRTUAL_***" to support more/most radios? Or can we just add "or event == EVT_DOWN_LONG" for T-Lite support?

kasatka60, 2021-12-29:
for Jumper T-lite:
[		elseif event == MENU or (not HORUS and  event == EVT_ROT_LONG) then](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/0b9da9888e6fe55f77eedcd983b692b0b858c1f8/src/SCRIPTS/TELEMETRY/iNav.lua#L520)
change:
elseif event == MENU or event == EVT_DOWN_LONG then

kasatka60, 2022-01-03:
How do I swap the up and down buttons in the menu?

stronnag, 2022-01-05:
@kasatka60  please try the Zip file in #26 (MDL long for menu).

kasatka60, 2022-01-06:
syntax error while switching screens (ent batton)
syntax error MDL long for menu

stronnag, 2022-01-09:
It would much appreciated if **single** issues for a **single** radio were raised per issue, otherwise it makes it very difficult for developers who don't own these radios.

* See https://github.com/iNavFlight/OpenTX-Telemetry-Widget/pull/26#issuecomment-1008316380 . Works in simulators for T12 and T-Lite. 
* As I don't have either radio, there is little more than can be done without more detail by enabling debug lua mode.

kasatka60, 2022-01-10:
jumper t12 pro opentx 2.3.14
Syntax error inav.lua:494 attempt to call a nil value

kasatka60, 2022-01-10:
I started this archive in the simulator and after that it worked. Only the up and down buttons in the menu are confused 
jumper t12 pro opentx 2.3.14

stronnag, 2022-01-10:
So let see if I understand this:

* On the T12 device you get the `Syntax error inav.lua:494 attempt to call a nil value` error (OTX 2.3.14)
* In the simulator (OTX 2.3.14), it works. Excellent.

I guess there could be some old file on the device that is not overwritten. Can you please try the following:
* remove  the `WIDGETS/iNav` directory
* remove the `SCRIPTS/TELEMETRY/iNav` directory
* remove the `SCRIPTS/TELEMETRY/iNav.lua` file
* Reinstall the inav widget from the Zip file
* Power cycle the device

On the matter of the keys,the script uses `EVT_VIRTUAL_INC` (Next) and `EVT_VIRTUAL_DEC`  (Previous) for navigation in uniformly across all radios. The mapping is  defined by OpenTX `radio/src/lua/api_general.cpp`,  e.g.
```
 { "EVT_VIRTUAL_DEC", EVT_KEY_FIRST(KEY_DOWN) },
 { "EVT_VIRTUAL_INC", EVT_KEY_FIRST(KEY_UP) },
```
If you feel this wrong for the T12, please raise an issue with OpenTX.

kasatka60, 2022-01-10:
I formatted the SD. Syntax error
After the simulator, the script works fine on the device. Except the up and down buttons.
Maybe I have a bad firmware.

stronnag, 2022-01-10:
I don't know, however:
* The key mapping is defined by OpenTX, not us
* The inav widget works on the simulator, that is all I can test

So I'll close this issue and merge #26 later today, which is the best I can do for t-lite and t-12.

stronnag, 2022-01-10:
Here's the final last shot. Please try the Zip below (on the radio). The only difference is that is it compiled on the desktop rather than on the device. It's often the case that in the simulator only "on-device" compiled lua will work; maybe the opposite applies to the radio?
[LuaTelemetry_v2.0.2-rc2.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/7839653/LuaTelemetry_v2.0.2-rc2.zip)

kasatka60, 2022-01-10:
Everything worked on the jumper t12 pro. As for the up and down buttons, I will figure it out myself. Everything worked correctly on older versions.

stronnag, 2022-01-10:
Excellent. We have changed nothing in regard to the buttons (as far as I'm aware).

closed 2022-01-10 by stronnag

kasatka60, 2022-01-10:
Jumper t-lite opentx 2.3.14 
Everything worked correctly. Only the up and down buttons in the menu are confused

2090570729, 2022-04-10:
Is it possible to set the required remote control interface on the computer to import the device?

2090570729, 2022-04-10:
此屏幕能否设置为默认打开

[image]

2090570729, 2022-04-10:
Can this screen be set to open by default

[image]

kasatka60, 2022-04-11:
Veiw mode: Pilot

kasatka60, 2022-04-11:
> Is it possible to set the required remote control interface on the computer to import the device?

Try to set it up in opentx/egdetx simulator.

2090570729, 2022-04-12:
In fact, the problem has been solved, and the interface can be switched normally? Thank you blogger,

2090570729, 2022-04-12:
In fact, the problem has been solved, the interface can be switched normally, thank you blogger,

kasatka60, 2022-04-15:
Do you mean "normally switch"?

2090570729, 2022-04-15:
Yes, it can be switched normally. I am using openTX2.3.14

2090570729, 2022-04-15:
Uploading video_20220415_230134.mp4…

2090570729, 2022-04-15:
Uploading video_20220415_231226_edit.mp4…

2090570729, 2022-04-15:
Uploading video.mp4…

2090570729, 2022-04-15:
I don't know, can you watch it?

nm17, 2022-04-15:
@2090570729 No, I can only see the text