iNavFlight/OpenTX-Telemetry-Widget #22: Up and down in the menu doesn't work NV14

Aaronsss, 2022-01-02:
### Describe the bug
I can open the options menu (left stick top right), I can select the option or exit the menu (right stick left / right). What doesn't seem to work is the scrolling up / down (right stick up/down) which means I can't select other options or change the current option in the menu

### To reproduce
Open the menu and attempt to move up and down through the menu on the NV14 running edgeTX

### Expected behavior
It should be possible to scroll up and down through the menu

### Radio and model settings
Radio: NV14
OS: Edge TX 2.6.0 RC1
Crossfire

stronnag, 2022-01-03:
Needs EdgeTX to fixethe simulator so this radio can be maintained.

nm17, 2022-01-04:
Tracking issue https://github.com/EdgeTX/edgetx/issues/1014 (I think)

stronnag, 2022-01-05:
@Aaronsss care to try the Zip in #25 to see if this fixes the issue please?

Aaronsss, 2022-01-05:
Closed via #25

Thanks for the quick fix

closed 2022-01-05 by Aaronsss