iNavFlight/OpenTX-Telemetry-Widget #27: Jumper T12, Jumper T-lite

kasatka60, 2022-01-15:
Jumper T12 edgetx 2.5.0, оpenTX 2.3.14
Jumper T-lite openTX 2.3.14 (Haven't checked edgetx yet )
LuaTelemetry_v2.0.2-rc2
Mixed buttons up and down in the menu inav telemetry. How to fix?

stronnag, 2022-01-15:
Raise an OpenTX issue,

closed 2022-01-15 by stronnag

konraddi, 2022-01-24:
Entering the menu worked in LuaTelemetry_v2.0.2-rc2 (via arrow button up. Arrow up and down are reversed when going through the menu) but now is no longer possible in LuaTelemetry_v2.0.2-rc3

Jumper T12 Pro

kasatka60, 2022-01-24:
I'll check later

stronnag, 2022-01-24:
Yeah, I  fixing the misuse of the previous misuse  of  PREV / PREVIOUS which stopped toggle min/max broke this on T12. We are running out of usable keys on these toy radios.

Looks like it's LONG press on  "Menu direction key" (12 on the manual diagram). Which is probably the sensible choice anyway. I'll put a test Zip out tomorrow.

I think t-lite is OK.

kasatka60, 2022-01-24:
Downloaded file LuaTelemetry_v2.0.2-rc4.zip. To enter the menu, long press the button **RIGHT** 
Jumper T12 Pro EdgeTX 2.6.0

kasatka60, 2022-01-24:
Jumper T-lite EdfeTx 2.6.0 is OK (except up and down keys)

konraddi, 2022-01-25:
Jumper T12 Pro with LuaTelemetry_v2.0.2-rc4 and opentx-t12 2.3.14-otx @stronnag @kasatka60

Entering the Menu now works via long press the arrow to the right. Arrows up and down are still reversed.

https://youtu.be/F7ry3YC0SCg

Many thanks,
Konrad

stronnag, 2022-01-25:
The inc and dec keys work as intended. Please state a version when they worked differently.

kasatka60, 2022-01-25:
> Jumper T12 Pro with LuaTelemetry_v2.0.2-rc4 and opentx-t12 2.3.14-otx @stronnag @kasatka60
> 
> Entering the Menu now works via long press the arrow to the right. Arrows up and down are still reversed.
> 
> https://youtu.be/F7ry3YC0SCg
> 
> Many thanks, Konrad

Works the same for me

konraddi, 2022-01-25:
> Whatever is done, some users will not like it, especially those who find the existing usage logical. Changing it It would also
> 
> * Affect every radio, not just this one
> * Require inverting the ugly stick logic for the colour radios.
> 
> Anyway, lets see whether there is overwhelming demand for change in #31

Tested 2.0.2-rc4 on FrSky Taranis X-Lite S with opentx-xlites 2.3.14-otx, entering the menu (via **Joystick down**) works too.

konraddi, 2022-01-26:
> > Whatever is done, some users will not like it, especially those who find the existing usage logical. Changing it It would also
> > 
> > * Affect every radio, not just this one
> > * Require inverting the ugly stick logic for the colour radios.
> > 
> > Anyway, lets see whether there is overwhelming demand for change in #31
> 
> Tested 2.0.2-rc4 on FrSky Taranis X-Lite S with opentx-xlites 2.3.14-otx, entering the menu (via **the button to the far right**) works too. Up and down joystick is inverted but is absolutely fine with me. Will check on my X9D+

https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/31#issuecomment-1021940281