iNavFlight/OpenTX-Telemetry-Widget #31: X9D, + and - buttons are inverted

Mateyhv, 2022-01-25:
On a Taranis X9D the function of the + and - buttons is inverted. (+) which is physicaly the upper button goes down the list and (-) which is the lower button moves up. Same behaviour on the OTX simulator.

stronnag, 2022-01-25:
Which version? Compared to which other version? Please.

stronnag, 2022-01-25:
> Which version? Compared to which other version? Please.

To answer my own question, this appears to be designed behaviour;  it works this way at least as far back as 1.7.4.
data.lua, line 126 etc.
```
		if event == EVT_VIRTUAL_INC or event == EVT_VIRTUAL_INC_REPT then -- Next option
```
The `INC` key does next (and the `DEC` key does previous).

Mateyhv, 2022-01-25:
This actually happened a few years ago but I believe Teckel repaired it. I cant tell about 1.7.4 as I have never used it in practice but if nothing has been changes probably the issue was there as well.

Mateyhv, 2022-01-25:
> > Which version? Compared to which other version? Please.
> 
> To answer my own question, this appears to be designed behaviour; it works this way at least as far back as 1.7.4. data.lua, line 126 etc.
> 
> ```
> 		if event == EVT_VIRTUAL_INC or event == EVT_VIRTUAL_INC_REPT then -- Next option
> ```
> 
> The `INC` (`+`) key does next (and the `DEC` (`-`) key does previous). The keys are not labelled UP and DOWN. Almost seems logical.

Its logical but not for the X9D where the buttons are used as UP and DOWN in all other menus. Right now pushing the down button goes up in the lua menu and viceversa.

konraddi, 2022-01-26:
> This actually happened a few years ago and I believe Teckel repaired it. I cant tell about 1.7.4 as I have never used it in practice but if nothing has been changes probably the issue was there as well.

On Version v1.7.4 it was fine with an older OpenTX version on my X9D+.

Since I upgraded OpenTX to 2.3.14, this older version has reversed button + and -

So it must be caused by some change in OpenTX 2.3.14 that also affects the 1.7.4. LUA version which was fine before...

stronnag, 2022-01-26:
I think you are correct. Prior to 2.3.14, the stick movement (up/dpwn) for editing  Inav lua config on colour radios (Horus, TX16, Nirvana etc.) was inverted. Then it miraculously changed to "normal", without the widget changing. It is possible that OTX changed stuff to rationalise LUA `EVT_VIRTUAL_*` across colour / roller / old style radios. OTX / ETX internally have greater 
visibility of key / roller events than LUA scripts such that the internal menus work intuitively (and there is per-radio f/w too).

I'm not sure that there is an easy  widget solution that fixes this for all radios without a lot of fragile "per radio" code (vice the "one size fits all" code we have now). I'm reluctant to do that, for maintainability reasons, at the moment.  But never say never ...

Mateyhv, 2022-01-27:
If I recall correctly it was something in OTX that changed the buttons behaviour and Teckel fixed it for the lua. Maybe OTX has reversed it once again. Closing this issue for cleaness.

stronnag, 2022-02-06:
This is symptomatic of the "grow like topsy" evolution of this project, and there being no (yet) consistent way of assigning capabilities to radios, rather there's a load of conditions based on name and / or screen  size; followed by exceptions where 'horus' type radios may or may not have a roller, and roller radios may not have a 'horus' screen. 

A future solution may be to assign a set of capabilities once based on known radio names and work from those; but that's for later. The immediate issue here is the way buttons are mapped to virtual events:

The non-roller radios have a mapping of VIRTUAL PREV/NEXT  for menu navigation) and INC/DEC (for incrementing / decrementing values);  these are effectively mirror images (`NEXT` == 'Down / Minus', `DEC` == 'Down / Minus') so they work for both cases (menu / editing values) .  The roller radios map PREV / NEXT and INC / DEC the same way, which doesn't work so well for two distinct cases.

| VIRTUAL Event | ROLLER | +/- | Up / Down |
| ------- | ------ | --- | --------- |
| PREV    | LEFT   |  +  | Up        |
| NEXT    | RIGHT  |  -  | Down      |
| DEC     | LEFT   |  -  | Down      |
| INC     | RIGHT  |  +  | Up        |

So here's a build that attempts to resolve the immediate conundrum within the context of current radio enumeration.
It would be much appreciated if @konraddi and @Mateyhv could give them a try and see if there is an improvement.
[LuaTelemetry_v2.0.2-rc6.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/8010584/LuaTelemetry_v2.0.2-rc6.zip)
[LuaTelemetry_v2.0.2-rc6_lua.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/8010585/LuaTelemetry_v2.0.2-rc6_lua.zip)

Mateyhv, 2022-02-06:
Tested, now buttons work as expected on a Taranis X9D+ with OpenTX 2.3.14. Thanks a lot @stronnag!

Mateyhv, 2022-02-22:
Ok, going to close the issue as it apparently is solved and to keep the tab clean. Thanks everyone!

closed 2022-02-22 by Mateyhv