iNavFlight/OpenTX-Telemetry-Widget #37: Radiomaster TX12 and other supported, but not listed in docs?

Matthias84, 2022-02-11:
### Describe your question
Hi, I love this script and the clever teamplay for inav and opentx. Unfortunatly I was pretty unsure, if my Radiomaster TX12 will be supported, as it's not listet, while other brands and models are explicitly listed. Thus I might decide to buy a more expensive modell or other brand, while this one working fine.

### Expected behavior

EIther list all supported radios, or generalise description that all OpenTX radios will work? (not sure which one is better / easier)

stronnag, 2022-02-11:
Please check the documentation before wasting developer's time with bogus issues. The TX12 is listed as a supported radio.
https://github.com/iNavFlight/OpenTX-Telemetry-Widget#features

closed 2022-02-11 by stronnag

Matthias84, 2022-02-11:
Sorry, I didn't saw it weeks ago. Thanks for pointing me there.

stronnag, 2022-02-11:
Please also see the 2.0.2 release candidates for supporting config menu access on the TX12.