iNavFlight/OpenTX-Telemetry-Widget #40: Widget chrashes TX16S into emergency mode on openTX 2.3.15

Arvycka, 2022-04-24:
Widget chrashes TX16S into emergency mode on OpenTX 2.3.15

When switching to the widget screen on the TX16S updated to OTX 2.3.15, radio crashes.
Trying to add the widget in a new screen also ends up in a crash.

### Radio and model settings
Radiomaster tx16s OpenTX 2.3.15

stronnag, 2022-04-24:
Possibly a bug in OTX then. A lua script should not crash the OS.
Works perfectly on EdgeTX.

stronnag, 2022-04-24:
In companion, a lua script should not be able to do this:
```
f_open(/home/jrh/Projects/TX16S/sdcard/SCRIPTS/TELEMETRY/iNav/horus.luac, a) = 0x7f4c2c16bc90 (FIL 0x7f4c29ff9fb0)
f_close(0x7f4c2c16bc90) (FIL:0x7f4c29ff9fb0)
convertToSimuPath(): /SCRIPTS/TELEMETRY/iNav/horus.luac -> /home/jrh/Projects/TX16S/sdcard/SCRIPTS/TELEMETRY/iNav/horus.luac
findTrueFileName(/home/jrh/Projects/TX16S/sdcard/SCRIPTS/TELEMETRY/iNav/horus.luac)
	found: /home/jrh/Projects/TX16S/sdcard/SCRIPTS/TELEMETRY/iNav/horus.luac
f_utime(/home/jrh/Projects/TX16S/sdcard/SCRIPTS/TELEMETRY/iNav/horus.luac) set mtime = Tue Jan 25 09:07:26 2022

luaDumpState(/SCRIPTS/TELEMETRY/iNav/horus.luac): Saved bytecode to file.
Segmentation fault

```

Arvycka, 2022-04-24:
Right, seems logical, same script works botx in otx 2.3.14 and edgetx. I'll report a bug on otx.

Arvycka, 2022-04-24:
OTX bug

stronnag, 2022-04-24:
If we are doing something wrong on OTX, then we would like to fix it. However, if OTX just crashes, it is impossible for this project to investigate further.

If you raise a bug on the OTX tracker, please cross-reference this issue so we can track it.

stronnag, 2022-05-01:
So there undocumented API changes in OpenTX 2.3.15
In the release note : No
In the API documentation: No

Update script later ....

stronnag, 2022-05-01:
#43 / 2.0.2-rc8  addresses the undocumented change to the OpenTX 2.3.15 `getVersion()` API.

closed 2022-05-01 by stronnag

Hankster1228, 2023-07-09:
Still not fixed.  I am running latest firmware, 2.3.15 on opentx, and still getting the EM when trying to update widgets with telemetry enabled.  Does not go into EM if telemetry disabled.