iNavFlight/OpenTX-Telemetry-Widget #41: EthOS

Teklock, 2022-04-25:
Any chance this LUA script can be made to run on FrSky's new line of radios running EthOS?

stronnag, 2022-04-25:
Unless 
* EthOS is API compatible with OTX/ETX; and
* FrSky wishes to supply the developers with sample hardware and API documentation

then there is little chance.

Teklock, 2022-04-25:
From what I understand, FrSky did supply hardware for the devs over at https://github.com/yaapu/FrskyTelemetryScript to port Yaapu to EthOS.

If you can, checkout https://www.rcgroups.com/forums/showthread.php?4018791-FrSky-ETHOS-Lua-Script-Programming and see how difficult it would be for you guys to make your wonderful script work with Ethos.  Thanks for the reply!

craigywsm, 2023-02-17:
I would be more than happy to donate if developers could get this to work

miggy1952, 2023-02-26:
I would also like this Lua for Ethos