iNavFlight/OpenTX-Telemetry-Widget #44: Inav Telemetry Log File

rickgode, 2022-05-15:
### Describe your question
In the inav lua script the ability to play log files seems to be greyed out.  Am I doing something wrong?

### To reproduce

1. Go to the inav menu and scroll down and try to click on the playback log
2. The playback log is greyed out
3. Should it be doing this?
[FPV Arrow-2022-05-14-125221.csv](https://github.com/
[image]
iNavFlight/OpenTX-Telemetry-Widget/files/8696053/FPV.Arrow-2022-05-14-125221.csv)

4. I think I have setup logging properly and will upload my logfile created from the radio

stronnag, 2022-05-16:
> In the inav lua script the ability to play log files seems to be greyed out. Am I doing something wrong?

I think this has been non-working for a while now. I'll take a look at it.

stronnag, 2022-05-16:
It would appear that at some stage the radio firmware (OpenTX probably, pre-Edge) changed the way log files are time-stamped, so the script never sees them.
This will require something of a redesign of the legacy log replay infrastructure, but possibly for the better.

stronnag, 2022-05-16:
@rickgode Firmware (Edge or Open TX) and version please.

stronnag, 2022-05-16:
So it would appear:
* OpenTX generated logs should still work, as they don't append the time to the log name
* It would be possible to have a separate log reader function for Edge (2.5 and later), as this supports a `dir(name)` iterator
  * Means having separate log handler for EdgeTX and OpenTX; ugly, but perhaps inevitable.

Otherwise there  is a workaround:
* Rename the log file to remove the time part of the name, e.g: `FPV.Arrow-2022-05-14-125221.csv` => `FPV.Arrow-2022-05-14.csv`

rickgode, 2022-05-16:
> EdgeTX v2.7.1
inav lua script v2.1.0

rickgode, 2022-05-16:
> So it would appear:
> 
> * OpenTX generated logs should still work, as they don't append the time to the log name
> * It would be possible to have a separate log reader function for Edge (2.5 and later), as this supports a `dir(name)` iterator
>   
>   * Means having separate log handler for EdgeTX and OpenTX; ugly, but perhaps inevitable.
> 
> Otherwise there is a workaround:
> 
> * Rename the log file to remove the time part of the name, e.g: `FPV.Arrow-2022-05-14-125221.csv` => `FPV.Arrow-2022-05-14.csv`
> * This will play, albeit exposing issues in the mapping between CRSF and Smartport used by the replay function.

> @rickgode Firmware (Edge or Open TX) and version please (EdgeTX I guess).

EdgeTX v2.7.1
inav lua script v2.1.0

rickgode, 2022-05-16:
> So it would appear:
>
> * OpenTX generated logs should still work, as they don't append the time to the log name
> * It would be possible to have a separate log reader function for Edge (2.5 and later), [...]

I changed the name of the log file like you said, but the Playback log is still greyed out

Thanks for looking into this.  Everything else seems to work fantastic, I just really wanted to see the playback animation on the map screen (especially in loiter mode with it circling around)

Ricky

stronnag, 2022-05-17:
> I changed the name of the log file like you said, but the Playback log is still greyed out
> 
> Thanks for looking into this. Everything else seems to work fantastic, I just really wanted to see the playback animation on the map screen (especially in loiter mode with it circling around)

Strange, it worked here (in the simulator at least). Not very impressive though, it doesn't display the motion and I don't see how it ever could have done from the code.

rickgode, 2022-05-17:
> 

I can't even get the inav screen to come up in my simulator...

stronnag, 2022-05-17:
Your log file in Companion / simulator from the INAV LUA log replay function (note the time, ignore the model name).
[image]

Your log file in Companion / simulator using Companion's log replay function. Note the location.
[image]

* In order to use INAV LUA in Companion / Simulator, it is often necessary to use the source archive from the release area, due to some mismatch between the firmware LUA interpreter and Companion's
* I didn't rediscover sensors, so maybe that's why some are missing
* Or more likely, Companion doesn't really understand the CRSF variant.

rickgode, 2022-05-17:
So I have the latest version of companion 2.7.1 and I used the source archive and now I can get the lua to show up in the simulator, but my log file is still greyed out.  

[image]

[image]

Thoughts?

stronnag, 2022-05-18:
Ah. It's a "feature" in Edge/OpenTX that we don't handle. 
* The model is called "FPV Arrow"
* ETX/OTX replaces the `space` with a `period` in the file name `FPV.Arrow-2022-05-14-125221.csv` (with time stamp) => `FPV.Arrow-2022-05-14.csv` (without).
* INAV LUA constructs  a file name from the verbatim model name and date (tries the last 15 days). 

The names then never match.

So the script needs to match the firmware name munging rules. Let's see if OTX/ETX use the same rules ....

stronnag, 2022-05-18:
It just gets better, in the legacy code we have:
```
local path = "/LOGS/" .. string.gsub(model.getInfo().name, " ", "_") .. "-20"
```
i.e. replace space with `_`. 
Please don't tell me OTX and ETX are gratuitously different here ... pretty please

stronnag, 2022-05-18:
## Findings

There are a number of issues here 
* OpenTX / EdgeTX replace the space differently
* Due to the naming conventions (`YYYY-MM-DD` v, `YYYY-MM-DD-HHMMSS`) we cannot easily predict  EdgeTX names (OpenTX just looks back 15 days). OpenTX does not have a LUA function to read directories, EdgeTX does).
* INAV LUA offers the last 6 (or fewer depending on time stamp) files for replay

## Solution

* Have separate log loaders for EdgeTX and OpenTX
* EdgeTX loader also handles plain `YYYY-MM-DD` (but not space substitution differences) 
* EdgeTX will also offer a maximum of 6 names from the last 15 days.

@rickgode Please try the contents of the attached Zip file (unzip over the extant files), perhaps in the simulator first. The file names are based on the SDCARD root. [edge_loader.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/8715524/edge_loader.zip)
```
Archive:  /tmp/edge_loader.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
    18067  2022-05-18 11:13   src/SCRIPTS/TELEMETRY/iNav.lua
     1299  2022-05-18 11:20   src/SCRIPTS/TELEMETRY/iNav/load.lua
     1431  2022-05-18 11:21   src/SCRIPTS/TELEMETRY/iNav/load_e.lua
     4428  2022-05-18 08:35   src/SCRIPTS/TELEMETRY/iNav/log.lua
    10106  2022-05-18 09:32   src/SCRIPTS/TELEMETRY/iNav/menu.lua
---------                     -------
    35331                     5 files
```

Note that at least for Smartport, the INAV LUA can replay available fields. If CRSF does not (no satellites, vehicle model doesn't move), then that's a **new Issue** please.
[image]

stronnag, 2022-05-18:
> Note that at least for Smartport, the INAV LUA can replay available fields. If CRSF does not (no satellites, vehicle model doesn't move), then that's a **new Issue** please. 

No need to bother. The legacy code decides to load the log player (Sport or CRSF) **before** it's determined whether the log is Sport or CRSF .... another PR once this one's done.

stronnag, 2022-05-18:
> No need to bother. The legacy code decides to load the log player (Sport or CRSF) **before** it's determined whether the log is Sport or CRSF .... another PR once this one's done.

[image]

* Space in name [Y]
* Recognised as CRSF [Y]

Some things are missing (fuel is because of a mismatch between telem units and lua configured unit), other units are inconsistently configured (speed, alt etc.). But that's just configuration.

Lack of actual movement / distance to be resolved.

stronnag, 2022-05-18:
Bingo!
[image]
@rickgode  Please confirm that the earlier ZIp at least fixes the loading, then I'll merge the whole lot.

rickgode, 2022-05-18:
Yay!  I think we're getting close!  The animation works on mine and it looks as it did while flying on the radio (exactly what I wanted)  All my sensors seems to be working as well (even the voice callouts) 

A coupe things.

1)  My flight mode doesn't seem to change it just stays on ACRO (I noticed your screenshot said cruise)
2)  I seem to have to rename it to FPV.Arrow-2022-05-14.csv so it still doesn't like the space so I renamed a couple other logs using the same convention like FPV.Arrow-2022-04-26.csv and it doesn't look like I can switch between.  I even took the 22-05-14 out of the LOGS folder and only left the 2022-04-26 file in there and the Playback Log  becomes greyed out again. 

We're getting so CLOSE!  The main reason I would like to get this going is so I can overlay these logs on my video since the blackbox doesn't seem to be working anymore on the Matek 405 minis https://github.com/iNavFlight/inav/issues/7719

Thanks for your help @stronnag 

[image]

stronnag, 2022-05-18:
> 1. My flight mode doesn't seem to change it just stays on ACRO (I noticed your screenshot said cruise)
> 2. I seem to have to rename it to FPV.Arrow-2022-05-14.csv so it still doesn't like the space [...]

1. My screenshot was Sport. We don't get flight mode in the Edge logs (at least not in the log of yours). This was (maybe is) also the case with OpenTX.
2. You've confused me now! The first example you sent me had the `period` replacing the space.

What exactly first appears on the SD Card on the TX for the log name ?
Space or period?

stronnag, 2022-05-18:
If you can produce a CRSF log where the `FM` field is populated (i.e. not all `0`) , flight mode can be implemented.

rickgode, 2022-05-18:
> 1. My screenshot was Sport. We don't get flight mode in the Edge logs (at least not in the log of yours). This was (maybe is) also the case with OpenTX.
> 2. You've confused me now! The first example [...]

Here is a video of exactly appears on the SD card 

https://user-images.githubusercontent.com/44140881/169117311-59dbd7fa-19c4-48b0-80ef-3bc47cbf90e6.mp4

FPV Arrow-2022-05-18-131842.csv  (I renamed the plane that's why the old logs just say FT Arrow.  I also made a video that might help explain a bit better.

rickgode, 2022-05-18:
> If you can produce a CRSF log where the `FM` field is populated (i.e. not all `0`) , flight mode can be implemented.

I gotta learn how to do this! Below is a video of it working, but when I take out the "." and put in a space it doesn't work.  Nor does the playback log seem to "see" the other files.

https://user-images.githubusercontent.com/44140881/169117486-f3382a85-726e-42fa-b480-e3dd016ee067.mp4

stronnag, 2022-05-18:
The videos really do nothing to help. Can you please:
* Remove all the old files from the LOGS folder
* Create a new log (for a model with a space in the name).
* Tell me what it's called. Do not change it, I'll change the script to suit.

stronnag, 2022-05-18:
Nevermind, I renamed one of my models on the TX16S to `Mr Spaced Out`
The log file is generated as: `LOGS/Mr Spaced Out-2022-05-18-184446.csv`

So it would appear no translation is needed for EdgeTX 2.7.1 at least.
How you got the `period`s remains a mystery (older EdgeTX maybe?)

stronnag, 2022-05-18:
https://github.com/iNavFlight/OpenTX-Telemetry-Widget/releases/tag/v2.1.1-rc1

I may in future think if we can support the `.` variant; however the above should address the 2.7.1 situation.

rickgode, 2022-05-18:
> The videos really do nothing to help. Can you please:
> 
> * Remove all the old files from the LOGS folder
> * Create a new log (for a model with a space in the name).
> * Tell me what it's called. Do not change it, I'll change the script to suit.

Sorry, I thought the videos would help in showing that even after edge 2.7.1 created the log file with a space in the name the inav lua script playback log still can't see them.

But... the v2.1.1-rc1 seems to allow the spaces and the time stamp as I can scroll the various log files that were created.  So working perfectly YAY!

Ricky

closed 2022-05-19 by stronnag