iNavFlight/OpenTX-Telemetry-Widget #47: Critical Battery alarm with a full battery.

OldManFPV100, 2022-06-08:
### Describe the bug
Hi , I am having a problem, every time I arm the motor I get a critical battery alarm, with a Full Battery, alarm stops when motor disarmed.
However, I do get a GPS lock, Ok to fly and the motor will spin. 
I have reset the flight and the telemetry, individually and together, prior to powering up the model and with model powered on.
The TX16s INAV LUA telemetry screen displays current Battery and Fuel at 100%, but I see it flashing yellow /red, 
Also, in the actual TX16s telemetry sensor screen I see the voltage fluctuating between 0-99%.

 I have followed the instructions in the WIKI R/T battery information and have matched the in INAV settings in configurator and the INAV lua on the TX16s I am using Crossfire, so I did not do the CLI changes as stated in the battery information. I did not have this problem until I tried changing over to EdgeTx then I changed back, after several failed attempts, everything else seems to working with the TX16s and the other models not associated with INAV.

Model #1
Ranger-FPV2022

Radiomaster Tx16s Running 2.3.15-otx
Crossfire nano RX
INAV config. 4.1.0
Matek 765 running INAV 4.1.0
2 x 5000mah 3 cell Lipo's

Thank You very much.
I hope it is something simple that I have missed!
Cheers

[RangerFPV2022.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/8857861/RangerFPV2022.zip)

stronnag, 2022-06-08:
If the value is fluctuation between 0-99% in the radio telemetry screen, then this is most likely a FC or RX/TX issue, not a widget issue; it's just showing what it gets from the radio.

Do you have an OpenTX CSV telemetry log that illustrates the affect?

OldManFPV100, 2022-06-09:
Hi Thank you for responding! I now have a file for the Open TX CSV telemetry log, I hope it helps.

[Ranger-FPV2022-2022-06-09.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/8865879/Ranger-FPV2022-2022-06-09.csv)
.

stronnag, 2022-06-09:
You have a FC/TX/RX telemetry problem. Nothing the LUA widget can do about this.
[image]

closed 2022-06-09 by stronnag

kyko84, 2022-06-12:
Hello, i have the same problem, here the telemetry log.
[Gekocrf-2022-06-12-183138.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/8886261/Gekocrf-2022-06-12-183138.csv)

stronnag, 2022-06-12:
Something is broken between the FC, RX and TX. The LUA widget is just the messenger.
[image]

kyko84, 2022-06-13:
I have a solution : 
The trick is to , turn off in the RX, the parameter "rx battery sensor".

https://www.reddit.com/r/fpv/comments/qbaufu/comment/hha615n/?utm_source=share&utm_medium=web2x&context=3

stronnag, 2022-06-14:
That makes sense, there is one useful field in CRSF telemetry for remote voltage (`RxBt(V)`), this is populated by INAV. If you also ask the RX to populate the same field, then each will overwrite the other at their respective update rates. This is why the graph oscillates between the external VBAT and the supply voltage to the RX. 

Configuration error.