iNavFlight/OpenTX-Telemetry-Widget #49: When I connected, got error(Script syntax error bad argument #1 to 'getV')

Addison666, 2022-07-02:


stronnag, 2022-07-02:
Care to provide some more information:
* Which OS and version (Edge, OpenTX etc.)
* Which TX model.

Addison666, 2022-07-02:
Open TX, MAMBO
version：TBS-1.3.0
Receiver：Tracer RX

stronnag, 2022-07-02:
versions of OpenTX are typically 2.3.15 etc. Is this an official OpenTX build?

Addison666, 2022-07-02:
> 

I think no official, maybe FW: fdtx-mambo

stronnag, 2022-07-02:
Please take the matter up with TBS "support" then; they can have our code, we don't have theirs.

* Not supported hardware
* Not supported firmware

closed 2022-07-02 by stronnag

Addison666, 2022-07-02:
> Please take the matter up with TBS "support" then; they can have our code, we don't have theirs.
> 
> * Not supported hardware
> * Not supported firmware

I checked version refer to official 1.2v0026

stronnag, 2022-07-02:
Which does not exist. Typical TBS, the Boris Johnson of support "obfuscate, deny, blame anyone else, deny, obfuscate...".
Please also check the list of supported OTX versions and hardware on our github page.

Addison666, 2022-07-02:
> 

The issue were happened when I had connected the receiver.
I import the script work well.

stronnag, 2022-07-02:
Still unsupported ...