iNavFlight/OpenTX-Telemetry-Widget #53: Can't go to config menu with literadio 3 pro, under the latest edgeTX

sikolenski, 2022-08-22:
### Describe the bug
the script works well, but there's no way to go in configuration menu. It seem like there'is no assigned button to do that.

stronnag, 2022-08-22:
Not a supported radio.

It is supported in the Edge Companion?

stronnag, 2022-08-25:
Once the radio is supported by a EdgeTx release (including in Companion), then we can considering supporting it. Please note that none of the developers have this radio and without Companion support, we cannot proceed.

sikolenski, 2022-08-25:
ok, thx for your answer

stronnag, 2022-11-26:
This should be working in the `lr3pro` branch with EdgeTX 2.8 lastest (probably also -rc4).
You can enter the menu by a **long press** on the right 'button':
[image]

@sikolenski , can you please try this archive **on the radio**?
[LuaTelemetry_v2.2.0-lr3.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/10096030/LuaTelemetry_v2.2.0-lr3.zip)

stronnag, 2022-12-11:
closing as fix is merged without any feedback from the OP

closed 2022-12-11 by stronnag