iNavFlight/OpenTX-Telemetry-Widget #54: Still getting nil value

j-keiter, 2022-08-22:
1) Be sure the OpenTX/JumperTX firmware on your transmitter includes the `luac` build option, video instructions linked here:
EdgeTX companion does not allow flashing so I cannot access the Lua build option.  OpenTX gives me a missing dfu usb device when I try to flash with the Lua build option.    I'm using Debian Linux.  Trying to get the OpenTX Lua Telemetry script to execute from the /Scripts/Telemetry folder on sd card.  EdgeTX 2.7.1 firmware on RadioMaster TX16S.  
2) When installing Lua Telemetry, be sure to download the latest LuaTelemetry.zip linked here:
I did this.
https://github.com/iNavFlight/LuaTelemetry/wiki/Installation
I did this.
3) Another possible reason for this error is that you're trying to use other Lua scripts on the same model. 
I'm not using any other LUA scripts on this model.  

4) Too many unused models. 
I deleted a few but  no change.  I only have 13 models.

Should I focus on trying to get OpenTX companion firmware write working such that I can flash the EdgeTX firmware with the LUA build option set?  I may have a permission issue concerning the no dfu device found error??

Thanks!

j-keiter, 2022-08-22:
When trying to flash via EdgeTX Companion 2.7.1.  The flash bar moves along but then I get:
Cannot write /media/jason/RM TX16S/FIRMWARE.BIN (reason: Unknown error)

stronnag, 2022-08-23:
You should open the widget from a telemetry screen, not from the filesystem.

j-keiter, 2022-08-23:
> You should open the widget from a telemetry screen, not from the filesystem.
At the 4:00 min mark in this video by RC Video Reviews, he executes the inav.lua script directly from the SYS - SDCard menu file structure.  That is why I thought it should work.  At 7:20 he mentions additional options when you run inav.lua as a script instead of widget.  That also motivated me to try and get it working.
https://www.youtube.com/watch?v=K8KZqkd29rg

stronnag, 2022-08-23:
If you want to do that, use the same unsupported, obsolete version of the script, which will not work on EdgeTX.

Running the contemporaneous script directly from the SD Card not a supported option, regardless of what old videos might show.

closed 2022-08-23 by stronnag

j-keiter, 2022-08-23:
So, current script is meant to be run only by the widget feature of EdgeTX and not by executing directly from Scripts folder on sd card.  Would be nice to have a test that warns user when trying to run as script.  Eg. "This script must be run as Widget" instead of nill value error.