iNavFlight/OpenTX-Telemetry-Widget #59: CRSF Transmitter Power not working

skaman82, 2022-09-15:
### Describe the bug
On CRSF the transmit power indicator stopped working. 
I am on CRFS FW 6.17 and OpenTX 2.3.15 on my TX16s

### To reproduce
- Create new model
- Discover Telemetry sensors
- Run the Telemetry Widget

### Expected behavior
Usually the widget is showing the transmit power of the CRSF module. 
I have another, older model on my radio (didn't refreshed sensor discovery on that) - there the mw Value is showing up normally - same FW and hardware.

### Screenshots
old model - no new sensor discovery
[image]

new model - with new sensor discovery. I also noted the name TWP2 is different - on the old model it is TWP.
[image]
[image]

skaman82, 2022-09-15:
Renaming the sensor manually into "TPWR" instead "TPW2" solves the issue

stronnag, 2022-10-28:
As this can be fixed by configuration rather than another exception in the code, I'm minded to close this.

closed 2022-10-28 by stronnag