iNavFlight/OpenTX-Telemetry-Widget #62: artificial horizon

effysh, 2022-12-01:
hello to all devs
I have a Horus 10s Accst with R9M reciver and modul
widget ver 2.2.0 
drone with  722 chip with betaflight 4.2 all parts of the widget works but not the artificial horizon .
also have a strange behave on horizon that flips upside down when ever I connect the drone to power blue sky down brown earth up
on betalight  configurator all is correct on 3d preview.
by the way older version which I had worked but I delete it and don't remember ver  num.

stronnag, 2022-12-01:
Please reproduce with INAV (and generate a TX log). Then I can look at it.

effysh, 2022-12-01:
> 

How do I access the TX logs?

stronnag, 2022-12-02:
You need to set a TX switch to enable logging (from "Special Functions"). In the screen shot (EdgeTX, TX16S), I've set 6 Pole switch position 6 to "SD Logs" at 0.2s interval (there are also audio prompts, but that doesn't matter). The log is written to the `LOGS/` directory on the SD Card.

Let's start with the BF log and take it from there. Please create a log that shows both the correct and incorrect state of the AHI.

Please also confirm which radio firmware you're using (EdgeTX or OpenTX) and its version number.

[image]

stronnag, 2023-01-02:
Closing, as no requested information provided.

closed 2023-01-02 by stronnag