iNavFlight/OpenTX-Telemetry-Widget #63: Battery level display error

PULLUUUUUUUUUUP, 2022-12-08:
Describe the bug
Battery level display error  on my TBS MAMBO and X9D 2019 (OpenTX 2.3.14)

Whether the problem is the flight controller?

To reproduce
The radio power on before the drone
Running the script
Percentage of battery is 0%
Battery voltage error

When The radio power on after the drone.The problem will be solved

Screenshots

stronnag, 2022-12-08:
* Not a supported radio
* Not supported firmware if it's the TBS fork 

* Works fine on supported radios / firmware
* Please contact TBS for support

closed 2022-12-08 by stronnag

PULLUUUUUUUUUUP, 2022-12-08:
The problem also occurred on X9D + 2019

stronnag, 2022-12-10:
* It does not happen on the maintainer's hardware. Thus it cannot be investigated further
* The script depends on the radio providing the available sensors. If the radio (reasonably) cannot do that if the model is not powered, then the script cannot show that data.
* Invoking the script and expecting it to work prior to powering the model is undefined behaviour.