iNavFlight/OpenTX-Telemetry-Widget #65: INAV LUA SCRIPT 2.2.0 - FREEZING In Radiomaster TX16S

Stemplar42, 2022-12-11:
### Describe the bug
Tengo un TX Radiomaster TX16S, modúlo ELRS Beta FPV de 1 Watts y el LUA INAV se freeza.

### To reproduce

1. Enciendo la radio
2. Enciendo el modelo con el RX
3. Toma toda la telemetría muy bien
4. A los 20 o 30 Segundos se freeza el LUA y no funciona más (incluso los audios), pero todos los mandos y el receptor funcionan bien.

### Expected behavior
El comportamiento esperado es que funcione todo como enlos primeros 20 segundos.
¿QUE PUEDE ESTAS SUCEDIENDO?

stronnag, 2022-12-11:
English please

wx4cb, 2022-12-11:
Describe the bug

I have a TX Radiomaster TX16S, a 1 Watt ELRS Beta FPV module and the INAV
LUA freezes.
To reproduce

   1. I turn on the radio
   2. I turn on the model with the RX
   3. It takes all the telemetry very well
   4. After 20 or 30 seconds the LUA freezes and doesn't work anymore (even
   the audio), but all the controls and the receiver work fine.

Expected behavior

The expected behavior is that everything works as in the first 20 seconds.
WHAT MAY BE HAPPENING?

stronnag, 2022-12-11:
Please provide the information requested in [the documentation](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Tips-%26-Common-Problems.md#reporting-issues-on-github), including a TX telemetry log of c. 5 minutes duration covering the loss of Lua functionality.

Stemplar42, 2022-12-11:
How do I get the Telemetry Log?. I have EdgeTX 2.8. Has anyone ever had this problem? Has anyone reported something similar?

stronnag, 2022-12-11:
No one has reported the problem, it does not happen on my TX16S, albeit S.port v. CRSF.

See https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/62#issuecomment-1334872410 for generating a log.

stronnag, 2023-04-17:
Closing as no requested information received

closed 2023-04-17 by stronnag

450quad, 2023-05-20:
Hi I installed a frsky receiver on my quad also have a TX16S
how/what  do I install next?