iNavFlight/OpenTX-Telemetry-Widget #66: iNav Telemetry on iFlight commando8

Cloudfan-official, 2022-12-23:
To Developers:
I want to run Telemetry Lua on iFlight commando8（EdgeTX 2.8.0）

But I can't enter the "Configuration menu", even if I press the "menu" button on the remote control.
Other operations are OK.

How can I solve this problem？
Thanks.

stronnag, 2022-12-23:
Please provide a link to the user guide. As the radio is supported by EdgeTX companion, it may be possible to enable the menu.

Cloudfan-official, 2022-12-23:
EdgeTX said that it support iFlight commando8 in v2.8.0:
https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0

What is your mean? user manual? or other things?
I've just started using "EdgeTX", and I still don't know something.

stronnag, 2022-12-23:
The user manual ...

Cloudfan-official, 2022-12-23:
[image]
[image]
[image]
[image]
[image]
[image]
They only gave such a simple user manual.

stronnag, 2022-12-23:
In that case it's not possible without the radio. Even then, with only two buttons, it may not be possible. 

Please contact the vendor for support.

closed 2022-12-23 by stronnag