iNavFlight/OpenTX-Telemetry-Widget #77: Inav telemetry lua radiomaster tx16s mode 1

aledwyngriffith, 2023-02-14:


stronnag, 2023-02-15:
The configuration menu is only available in mode 2. 
There is no other way to access the configuration on colour radios.

If you're prepared to do some text file editing, you can edit the model's configuration file directly (on the SD Card).

The file is (from the root of the SD Card ) is `SCRIPTS/TELEMETRY/iNav/cfg/MODELNAME.dat` (where MODELNAME is the name of model). The file is a fixed length text string of 47 characters. This is a little arcane; description below:

The number of characters for each item is defined in `SCRIPTS/TELEMETRY/iNav/config.lua` and the acceptable values in `SCRIPTS/TELEMETRY/iNav/menu.lua`. 

Here's a example:
```
13534230120011121110020302035003000145004200000
```
So lets annotate this a bit to show positions:
* First line is the actual data
* Second line is "tens"
* Third line is "units" . The asterisk marks the position we want.

```
13534230120011121110020302035003000145004200000
0        1     *   2         3         4
12345678901234567890123456789012345678901234567
```
From `config.lua`, we can see `Battery Alert` is the 16th character; from `menu.lua` we can see it takes values of `0`=None, `1`=Critical, `2` = All (so in the example, `2` means all).

Change the `2` to `0` and you should no longer have battery alerts 

* Make a backup in case you mess it up (or the above instructions are wrong)
* Only save the configuration 47 bytes, no line-feed, no additional lines.

aledwyngriffith, 2023-02-15:
Thanks so much for taking time to reply to my query. Fascinating insights
into how these systems work! I  remain primarily concerned about the fact
that I cannot discover the vbat sensor on the f411wte flight controller and
assume this is the reason the voltage field in the telemetry screen is not
populated. This occurs in 2 builds with this hardware combination probably
eliminating fc being faulty as the cause. I have tried various inputs in
the voltage and current sections of inav configurator with no success. The
most logical would appear to be adc raw with factors of 1100 and 250 for
voltage and current respectively as suggested in the fc documentation.  I
don't want to incur the expense or time investment adopting another flight
controller when the problem might be a software issue.
Thanks again for your interest.
Regards,  Aled Griffith

50UR4V, 2023-02-15:
> Radiomaster tx16s edge tx version 2.8 latest version Fr sky x8r with s port telemetry Inav 5.1 on matek f411wte Inav lua script 2.2 latest version Unable to access widget configurator. stick movements don't work on mode 1 (throttle right) is there another way to access the configuration? Voltage not displayed, unable to see vbat sensor. Other sensors detected no problem. Same problem with 2 builds using the same hardware. Results in constant low battery warning which I have no access to configuration to turn off. Re setting telemetry no use. Be grateful for any help.

Try this:
1. Make sure you are in the inav telemetry widget screen.
2. Raise the throttle (right stick) to top. Full throttle.
3. Move rudder (left stick) to right. Full right rudder.
4. You should be in the config menu. Move throttle to zero to start editing.
5. Elevator (left stick) to cycle through menu/settings.
6. Aileron (right stick) to select and deselect.

It's exactly the same instructions mentioned in https://luatelemetry.readthedocs.io/en/latest/Configuration-Settings/ but with the sticks mapped to mode 1.

I have always used this method on my TX16S running inav lua 2.2. I am right now on inav 6.0 FP2 (pretty sure that doesn't matter)

stronnag, 2023-02-15:
> It's exactly the same instructions mentioned in https://luatelemetry.readthedocs.io/en/latest/Configuration-Settings/ but with the sticks mapped to mode 1.
> 
> I have always used this method on my TX16S running inav lua 2.2. I am right now on inav 6.0 FP2 (pretty sure that doesn't matter)

Yeah. That even works in Companion (Mode 1).

aledwyngriffith, 2023-02-16:
Thanks once more.  This works flawlessly!
Needs to be published in the documentation.
Sadly, still not discovering a vbat sensor on my f411wte  fc. Vfast is
present but battery voltage not populated on the inav lua. /-:
Cheers for your help,
regards, Aled

closed 2023-04-17 by stronnag