iNavFlight/OpenTX-Telemetry-Widget #82: Syntax error on RadioMaster Boxer

jimsynz, 2023-03-12:
Hi folks!

### Describe the bug

I'm seeing a syntax error when attempting to run the widget (both the latest release and master):

```
Syntax error
?:0
attempt to index field '
table' (a nil value)
```

This happens regardless of whether a model is connected or not.

My guess is that it's actually a memory-related issue, but I'm logging it here in case other folks also run across it.

My setup:

- RadioMaster Boxer (ELRS version)
- ExpressLRS 3.2.0
  - Packet rate: 100Hz Full
  - Telem ratio: 1:2
  - Switch mode: 12ch mixed
  - Model match: off
  - TX power: 1000mW Dyn
- No other lua scripts running
- Only a single model, named "INAV"

### To reproduce
(with the INAV widget configured as a telemetry screen)

1. Press "TELE" button

### Expected behavior
I should see the telemetry widget.

### Radio and model settings
This zip file contains only the `etx` file as the `cfg` directory is empty on my radio.

[boxer.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/10952218/boxer.zip)

jimsynz, 2023-03-12:
Agreed, but at least it's searchable now.

lanyiuav, 2023-03-17:
so，how to Slove thequestion

stronnag, 2023-03-18:
> so，how to Slove thequestion

* Test it on the EdgeTX nightlies, if it works, wait patiently for EdgeTX 2.9
* Otherwise, send  (or persuade the manufacturer to send) a Boxer (TX, relevant RX) to the maintainer in order that the issue may be investigated.

prelator, 2023-04-15:
I am seeing this error on my Boxer radio as well. Telemetry worked at first while I was ground testing. On my first flight with this plane, the widget just showed invalid numbers for everything, but audio alerts worked though the battery ones were completely wrong. Then I turned on telemetry logging on my radio and changed the battery alert levels in the lua script, and then next time I went to fly it gets this syntax error every time I try to run the lua script. Seems this script simply isn't compatible with the Boxer radio.

stronnag, 2023-04-17:
Fixed by #86 (probably)

jimsynz, 2023-04-17:
Hi there.

Just pulled master and copied it onto the boxer, now when switching to the telemetry widget I see the following:

```
Syntax error
iNav.lua:541
attempt to call a nil value
```

prelator, 2023-04-17:
I'm seeing the same thing and commented about it on the related ticket. The answer was just that the Boxer isn't supported. Shame since it actually was working fine on my Boxer as long as I didn't have any saved log files. With this change it just won't work at all.

jimsynz, 2023-04-21:
I've just upgraded to EdgeTX 2.8.3 and now I get the following new error:

```
Syntax error
?:0
attempt to call a field 'getElrsRFM' (a nil value)
```

what's new is that you can actually see the UI behind it (whereas before I just saw a blank screen).

[image]