iNavFlight/OpenTX-Telemetry-Widget #84: Throttle warning with flaps on throttle stick

ucubeco, 2023-04-15:
Radiomaster TX16S Expresslrs 3.2 and Edgetx version 2.8, mode 1
Radiomaster EP3 Expresslrs 3.2
Inav 3.0.2 on Matek F722 WPX
Inav lua script 2.2

I'm configuring a F5J sailplane where I mapped throttle on the right slider and flaps on the throttle stick. Flaps zero is stick full up. Now I'm getting a throttle warning in iNav lua while flight status is 'not ok'. As soon as a GPS fix is available and the status changes to 'ready', the warning is gone. So it's not really a problem, but rather somewhat annoying.

It seems to me that to generate the warning, always the throttle stick position is read regardless of the channel mapping. Is there any fix for this?

stronnag, 2023-04-15:
The widget reports the status provided by INAV. This looks like normal INAV arming status based on the requirement for a GPS fix because you have some navigation function configured.
The widget does not read the sticks.

closed 2023-04-15 by stronnag

ucubeco, 2023-04-16:
I found the code that causes my issue: The stick position is read in iNav.lua line 148:

`data.throttle = getValue(data.thr_id)`

And the throttle warning is generated in line 234:

`data.modeId = (bit32.band(modeE, 2) == 2 or modeE == 0) and (data.throttle > -920 and 12 or 5) or 6 -- Not OK to arm(5) / Throttle warning(12) / Ready to fly(6)`

I changed this line to:

`data.modeId = (bit32.band(modeE, 2) == 2 or modeE == 0) and 5 or 6 -- Not OK to arm(5) / Ready to fly(6)`

which fixed my problem :)