iNavFlight/OpenTX-Telemetry-Widget #86: Error ?:0 attempt to index filed table

druckgott, 2023-04-16:
### Describe the bug
I have a Taranis X-Lite Pro and flashed EdgeTX "Flying Dutchman" v2.8.2 and Update the Telemetry Widget to the newest release Release 2.2.0
https://github.com/iNavFlight/OpenTX-Telemetry-Widget/releases/tag/v2.2.0
Bevor I had an old version 1.7.5 which was quite old but was running as Widget on EdgeTX "Flying Dutchman" v2.8.2

### To reproduce
Update Widget from Version 1.7.5 to 2.2.0

### Expected behavior
No Error

### Screenshots
Syntax error
?:0 attempt to index field 'table' (a nil value)

### Radio and model settings
Taranis X-Lite Pro with FrSky R9M Lite Pro Modul 900mhz

[Chimera7_dat.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242525/Chimera7_dat.zip)
[document1.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242527/document1.zip)

stronnag, 2023-04-16:
AFAIK, there seem to be a number of issues in ETX 2.8 with `table`  (e.g.  https://github.com/EdgeTX/edgetx/pull/2994#issuecomment-1415460802) which are scheduled for fixing in EdgeTX 2.9.

stronnag, 2023-04-16:
But maybe it's worse than that, the `table` API is only available on colour radios:
```
$ less edgetx/radio/src/thirdparty/Lua/src/linit.c
...
#if defined(COLORLCD)
  {LUA_TABLIBNAME, tab_funcs, NULL, NULL},
#endif
```
So we will have to disable log replay on all non-colour radios ...

druckgott, 2023-04-16:
If I comment out in load_e.lua the error change. So maybe it´s in this area?!

```
-- From EdgeTX 2.7.1 (at least) we don't need to translate spaces
local mbase = model.getInfo().name .. "-20"
local mblen = string.len(mbase)
local tempf = {}
local tempi = 0
for fname in dir("/LOGS") do
   if string.find(fname, mbase, 1, true) == 1 then
      local dstr = string.sub(fname, mblen+1, -5)
      local lyr = tonumber(string.sub(dstr, 1, 2))+2000
      local lmo = tonumber(string.sub(dstr, 4, 5))
      local lda = tonumber(string.sub(dstr, 7, 8))
      tdiff = 366*(log.year-lyr) + (log.mon-lmo)*31 + log.day - lda
      if tdiff < 16 then
         tempi = tempi + 1
         tempf[tempi] = dstr
      end
   end
end
```

stronnag, 2023-04-16:
Yes, I know exactly where it is ...

stronnag, 2023-04-16:
So by "error change" do you mean the problem disappears or it changes to another problem?

druckgott, 2023-04-16:
No only a other error massage is comming up ;-) So that´s why I mabye think it´s this area:

I get:

```
load_e.lua:48
attempt to compare number with nil
```

but maybe I´m on a wrong way.

stronnag, 2023-04-16:
No your investigation helps, thanks. I will post something for you to test "soon"

druckgott, 2023-04-16:
Ok mega strange:

I comment out the load_e.lua: like this:

```
-- Log loader for EdgeTX

local config, data, FILE_PATH = ...

-- Load config for model
fh = io.open(FILE_PATH .. "cfg/" .. model.getInfo().name .. ".dat")
if fh ~= nil then
   for i = 1, #config do
      local tmp = io.read(fh, config[i].c)
      if tmp ~= "" then
	 config[i].v = config[i].d == nil and math.min(tonumber(tmp), config[i].x == nil and 1 or config[i].x) or tmp * 0.1
      end
   end
   io.close(fh)
end

-- Look for language override
fh = io.open(FILE_PATH .. "cfg/lang.dat")
if fh ~= nil then
   local tmp = io.read(fh, 2)
   io.close(fh)
   data.lang = tmp
   data.voice = tmp
end

local log = getDateTime()
config[34].x = -1

-- From EdgeTX 2.7.1 (at least) we don't need to translate spaces
--[[local mbase = model.getInfo().name .. "-20"
local mblen = string.len(mbase)
local tempf = {}
local tempi = 0
for fname in  dir("/LOGS") do
   if string.find(fname, mbase, 1, true) == 1 then
      local dstr = string.sub(fname, mblen+1, -5)
      local lyr = tonumber(string.sub(dstr, 1, 2))+2000
      local lmo = tonumber(string.sub(dstr, 4, 5))
      local lda = tonumber(string.sub(dstr, 7, 8))
      tdiff = 366*(log.year-lyr) + (log.mon-lmo)*31 + log.day - lda
      if tdiff < 16 then
	 tempi = tempi + 1
	 tempf[tempi] = dstr
      end
   end
end
]]--

if tempi > 0 then
   table.sort( tempf )
   for i = #tempf, 1, -1 do
      config[34].x = config[34].x + 1
      config[34].l[config[34].x] = tempf[i]
      collectgarbage()
      if config[34].x == 5 then break end
   end
end
collectgarbage()
return
```
I get an error as one post above.

I get back and change the file back to the original.

And now I get a working screen with no error?! Whats happening?!!??!?! Very stange.

stronnag, 2023-04-16:
Do you have a `LOGS` directory (on the SD Card) and does it have any files in it?

druckgott, 2023-04-16:
I Update my post bevor with some more informaations to your questions because I realized it parallel ;-) same what you think ;-)

stronnag, 2023-04-16:
You answered my question. Deleting all the files will fix it. However, the test is not conclusive, as copying the `.lua` file to the radio will have caused a new `.luac` to be compiled on the radio. It could also be the case that the precompiled (on a desktop computer) `*.luac` files are not compatible with EdgeTX's lua interpreter.

Can you please do the following: with the replaced `lua` file, put the logs back and see what happens

druckgott, 2023-04-16:
1. Ok with Log files restored an original File I get the error again. (Error)
2. With removing load_e.luac and comment out like aboce (Error)
3. With removing load_e.luac and original File and no log files (no Error)

druckgott, 2023-04-16:
I also get the error on companion If I use my SD Card content

Here are some of my log files:

[image]
[Chimera7-2022-02-20.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242677/Chimera7-2022-02-20.csv)
[Chimera7-2022-02-19.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242678/Chimera7-2022-02-19.csv)
[cineWhoop-2021-01-30.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242679/cineWhoop-2021-01-30.csv)
[cineWhoop-2021-02-03.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242680/cineWhoop-2021-02-03.csv)
[IFlight5Z-2021-04-22.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242682/IFlight5Z-2021-04-22.csv)
[IFlight5Z-2021-05-14.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242683/IFlight5Z-2021-05-14.csv)

stronnag, 2023-04-16:
Thanks. I'll remove the ability to enumerate / replay logs on non-colour radios and post some test archives here "soon".

stronnag, 2023-04-16:
Care try these ?

* Pre-compiled `.luac` : [LuaTelemetry_v2.2.0+log.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242893/LuaTelemetry_v2.2.0%2Blog.zip)
* Compile on TX `.lua` : [LuaTelemetry_v2.2.0+log_lua.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11242888/LuaTelemetry_v2.2.0%2Blog_lua.zip)

stronnag, 2023-04-16:
Thanks for confirming. Maybe this will also help #82

closed 2023-04-17 by stronnag

prelator, 2023-04-17:
I tried using the files posted by stronnag above hoping it would fix the issue on my Radiomaster Boxer. However now when I try to run the script I get the following error:

`Syntax error: iNav.lua:541 attempt to call a nill value`

Is there anything else I need to do to get this working? I had deleted all the iNav telemetry files from my radio and replaced them with the files from the zip above. I tried both the precompiled version and the compile on radio version and got the same result.

stronnag, 2023-04-17:
* Boxer is not a supported radio
* The maintainer does not have a Boxer radio; alas, manufacturers prefer to send samples to "influencers" rather than developers who actually maintain products that run on their hardware
* The script runs perfectly in the Companion / Simulator

prelator, 2023-04-17:
Yeah I know it's not supported, but the script was formerly working for me as long as I didn't have any log files in my logs directory. I had hoped fixing that issue would allow this to work on my radio.

prelator, 2023-05-13:
I tested the latest version of the telemetry widget on my Boxer and it appears to work now even when logs are present. Hopefully this continues to work with no more errors.