iNavFlight/OpenTX-Telemetry-Widget #96: Syntax error ?:0 attempt to call a field 'getElrsRFM' (a nil value)

druckgott, 2023-04-29:
### Describe the bug
I have a Taranis X-Lite Pro and flashed EdgeTX "Flying Dutchman" v2.8.2 and Update the Telemetry Widget to the newest release Release 2.2.0
https://github.com/iNavFlight/OpenTX-Telemetry-Widget/releases/tag/v2.2.0
Bevor I had an old version 1.7.5 which was quite old but was running as Widget on EdgeTX "Flying Dutchman" v2.8.2

I also have Expresslrs the current version 3.2.1, I switched this today from frsky, bevor the script works perfekt

### To reproduce
switch from frsky to express lrs.
On call of lua widget:
```
Syntax error
?:0
attempt to call a field 'getElrsRFM' (a nil value)
```

### Expected behavior
no error

### Screenshots

### Radio and model settings
Taranis X-Lite Pro with Radiomaster Ranger Nano Modul + RP3 Radiomaster reciver (also happens with RP1/RP2)

stronnag, 2023-04-29:
Looks like a typo in the big RFMD changes, can you please unzip the attached and install the new `SCRIPTS/TELEMETRY/iNav/func_t.lua` on the radio (and delete the  extant  ` SCRIPTS/TELEMETRY/iNav/func_t.luac` for good measure), reboot and see if that fixes things?
[func_t_lua.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11359465/func_t_lua.zip)

druckgott, 2023-04-29:
its working thx

stronnag, 2023-04-29:
Thanks for confirming. TIme for 2.2.1  then

PR #97 (2023-04-29): fix typo in BW ERLS function
Should fix #96
closed 2023-04-29 by stronnag