iNavFlight/OpenTX-Telemetry-Widget #99: The script is not working with my system. ELRS + iNav + QX7

0Balazs, 2023-05-29:
### Describe the bug
The script was worked perfectly with my old rx-tx module (QCZEK LRS) but i bought a new Betafpv 868 elrs system. I am getting the necessary telemtry datas, but an another ID with the Qczek. So the Ianv lua didnt see the datas, only the Voltage and current. But all of the datas is availabke on the telemtry settings screen.

### To reproduce

### Expected behavior
The script has to see all of the datas and working correkt. Because elrs is using crsf if iam correct. And crsf is supported.

### Screenshots

### Radio and model settings

### Additional context

stronnag, 2023-05-29:
Please follow the documentation for providing necessary data to investigate the issue, in particular a telemetry log.
https://luatelemetry.readthedocs.io/en/latest/Tips-And-Common-Problems/

0Balazs, 2023-05-29:
[RAngerelrs-2023-05-29.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/11591049/RAngerelrs-2023-05-29.csv)

Sorry. There are a telemetry log. I hope it will help.

stronnag, 2023-05-29:
I don't see any script issue. Using the Companion, I see on the sensors page (bogus) data.
In the script I see the same bogus data.  The script reflects exactly the data on the sensors page.

Note I do not have any of this hardware, so unless someone else volunteers, this issue cannot progress.

0Balazs, 2023-05-29:
Ok, so my friend has a same setup and a same problem. Some data is correct (but blinking, even if we are using higher packet rate). But the artificial horizont is almost in the deafult position.
I am trying read the program but i am new in lua. I am getting the pitch and roll datas are the same ID (001E) but different sub ID (1 and 2). But the name is correct (Roll and Ptch).

haklis, 2023-05-29:
Hi, I have the same problem. Here are some details:
- New ELRS system (Ranger micro) on a Taranis X9D+, RX/TX flashed to latest 3.x.x
- Mateksys 765-Wing
- INAV 6.1
- OpenTX-Telemetry-Widget/releases/tag/v2.2.1
- 24 sensors are discovered and are updated frequently (several times a second for some of them)
- Telemetry widget opens on the radio when connected to inav without error messages, but the values don't populate and are blinking.

What can I do to give you more debug info?

stronnag, 2023-05-30:
This is due to a bug in OpenTX. It does not populate the FM (Flight Mode) `UNIT_TEXT` field correctly. 
This problem does not exist in EdgeTX ([Issue](https://github.com/EdgeTX/edgetx/issues/1991) and [Edge fix](https://github.com/EdgeTX/edgetx/pull/2173) ). 

An OpenTX [issue](https://github.com/opentx/opentx/issues/6413) was raised for four years ago and closed without being fixed.

Upgrade to EdgeTX and it will work. It cannot be fixed here.

closed 2023-05-30 by stronnag

haklis, 2023-05-30:
Thanks for the feedback, I will switch over to edge tx. It would be nice however if the documentation specifically mentioned that opentx is not supported with elrs - if it is that combo that causes the bug. 😀

0Balazs, 2023-05-31:
Ok, so we tested it with the latest EdgeTX FW, and it is working now.

Thank you.