iNavFlight/OpenTX-Telemetry-Widget #103: Jumper T-Lite - Not able to enter configuration mode.

Kikerumbo, 2023-07-25:
Greetings,

I've been trying to configure the alarms (I use Li-Ion and the alarms are set too high) and It's not possible to enter the configuration with a Jumper T-Lite with EdgeTX.

In the guide is stated that you should long press the "MENU" button to enter, but none of the buttons on the radio get me into the configuration menu of the LUA.

Can this issue be solved or can I configure the alarms in the LUA any other way?

Thanks.

stronnag, 2023-07-25:
Works fine in the Simulator. Long press on the lower right button (whatever that may be).

[image]

Kikerumbo, 2023-07-25:
> Works fine in the Simulator. Long press on the lower right button (whatever that may be).
> 
> [image]

Thanks for the answer.

The problem is still the same. Here I link a video of the issue (trying what you told me, even if I already tried before I posted). Lower right button is "MDL".

https://youtube.com/shorts/nW9xRd1KelA?feature=share

stronnag, 2023-07-26:
Whatever, the code uses the firmware's `EVT_VIRTUAL_MENU_LONG` for the menu. If this is different in the simulator and the hardware, then it is a EdgeTX or vendor issue. Please contact them for help. The maintainer does not have access the hardware.

closed 2023-07-26 by stronnag

stronnag, 2023-07-26:
See also #106

Kikerumbo, 2023-07-27:
I did what I could to help, showing the issue. As I see the information isn't useful to solve a problem that does only exists with my radio, I apologize.

Since it seems I have to be "a programmer" to test why a LUA script doesn't work (I don't understand the mwptools installation guide), I'll just use the other script (GPSx9L).

Thanks for the "help".

You can re-close the issue.

caiuspetronius, 2023-08-24:
The same problem with Commando 8.

caiuspetronius, 2023-08-24:
Looks intriguing. And how exactly did you achieve the configuration menu on Commando 8? It took me an hour of experimenting to figure out that a long left push on the left joystick switches the screens view ).

caiuspetronius, 2023-08-24:
I just edited the confit.lua script by hand for now.

stronnag, 2023-08-24:
I followed the [documentation](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Configuration-Settings.md),  maybe you should try it.

caiuspetronius, 2023-08-24:
The link you provided has a newer version of the docs than what I was reading. There was nothing there about the new script to test buttons. Anyway, the script shows that long MDL (long right on the left joystick)sends the EVT_VIRTUAL_MENU_LONG event, but it doesn’t work for me nevertheless. I’m trying it from the screen where the iNav telemetry script is displaying its graphics. Is this the right location?