iNavFlight/OpenTX-Telemetry-Widget #111: Telemetry Stopped After Upgrading TX from OpenTX to EdgeTX

NateTL, 2023-08-12:
I am running V2.2.1 (had same problem with previous versions as well) on a Jumper T16 running EdgeTX 2.7.0 and a R9M TX module.  The LUA widget is not showing any telemetry despite the fact it shows up fine in the model settings Telemetry page.  Over a year ago I upgraded from OpenTX to EdgeTX and this is when it stopped working.  I tried many things to get it working and then just gave up.  I have three different fixed wing aircraft and they all have the same issue.  I am currently running iNav v6 but had the same problem on previous versions (V4 and V2). 

I guess my first question is whether this specific combination of T16, EdgeTX and a R9M should be working?  Does anyone else have it working?  If the answer is yes, I'm wondering what I might try to get this working?  I am able to create a standard widget to display things like distance to home, altitude, GPS co-ordinates, flight battery voltage, etc. but would much prefer to use the LUA telemetry script.  Thanks.

### Screenshots
Here are some screenshots showing the telemetry working in the model setup and a regular widget and one showing it not working in the LUA script.
[OneDrive_1_2023-08-12.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12328020/OneDrive_1_2023-08-12.zip)

### Radio and model settings
The EdgeTX models and .dat file are attached.  The models that have this problem are called Dart 250g, Nano Talon and Nano Goblin.
[LUA-Files.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12328014/LUA-Files.zip)

stronnag, 2023-08-12:
Please provide EdgeTX telemetry logs.

NateTL, 2023-08-12:
Here is a log from the Dart 250g.
[Dart 250g-2023-05-27-102703.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12328084/Dart.250g-2023-05-27-102703.csv)
Here is a more recent log from another of the planes with the issue:
[Nano Goblin-2023-08-05-110705.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12328088/Nano.Goblin-2023-08-05-110705.csv)

stronnag, 2023-08-12:
Thanks. I had a brief look at the Nano Goblin log. This indicates some issue with telemetry as seen by the radio, for example:

* RSSI(dB) values > 5500
* Multiple instances of Galt(m), with different values
* AccZ(g) of 5516. I doubt your model can withstand 5516 g.
* Multiple instances of Tmp1(@C) with different values (INAV and INAV Lua encode  flight mode data in this field).

This bizarre log indicates that the problem is outside of the Lua Widget.

closed 2023-08-12 by stronnag

NateTL, 2023-08-12:
Not sure why that log file is wonky.  I looked at a few others and they all have more realistic values and I have the problem with them as well.  Here is another one (converted to .xlsx as I find them easier to work with).  Does it look any better or do you see things in there that would mess with the LUA script?  Thanks for your time in looking into this.

[Nano Talon-2022-06-25-085542.xlsx](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12328191/Nano.Talon-2022-06-25-085542.xlsx)

stronnag, 2023-08-12:
Converting logs to proprietary formats really does not help ...

NateTL, 2023-08-12:
Sorry, here is another in CSV...
[Nano Talon-2022-08-20-083428-CRASH.csv](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12328202/Nano.Talon-2022-08-20-083428-CRASH.csv)

stronnag, 2023-08-12:
I had a look at the Dart log. It seems sane
[image]

It replays in the Companion / Simulator into the INAV Lua just fine too.

[image]

stronnag, 2023-08-12:
[image]

NateTL, 2023-08-12:
Yes, I've looked at them on Google Earth before so figured the problem isn't with the telemetry data.  Just can't figure out why it doesn't display anything on the LUA script widget.   Any idea what might be the issue or things I could try.  Thanks again.

stronnag, 2023-08-13:
The widget is just a messenger for the data obtained via OTX/ETX APIs. In order to display in simplistic  terms it requires "telemetry", i.e. RSSI > 0 and for flight data that the FC is armed.

The reasonable logs both replay as expected via the Companion; I would suspect some corruption of the models from an earlier OpenTX to EdgeTX to conversion (this was not perfect in early ETX and then again with the binary to YAML transition.

If it were my issue, I'd:
* Update to the latest ETX (2.8.5)
* Start again with a new, minimal model and once you've validated that, add in the extant bells and whistles.

NateTL, 2023-08-15:
OK, thanks for the suggestions - will try both of those and see what happens.