iNavFlight/OpenTX-Telemetry-Widget #113: 2.2.1 breaks Crossfire Compatibility on X9D+ Radios

b14ckyy, 2023-08-15:
### Describe the bug
With the Version 2.2.1 on the X9D+ Radio and Crossfire the Widget shows lost telemetry alarm constantl. It still Updates certain values like Battery Voltage, Current Draw, Sat count and maybe some more but not the Attitude, GPS coordinates and constant Telemetry Lost Warning. Works fine on TX16S with Crossfire.
Downgrading to 2.2.0 solves the issue. 

### To reproduce
1. Install 2.2.1 on a X9D+ or similar
2. Bind a Crossfire INAV Craft and discover sensors
3. Restart Radio to refresh the LUA Sensor detection
4. Check LUA status

### Expected behavior
LUA Should work normal also with Crossfire

### Screenshots
[image]

### Radio and model settings
No Special Settings or config needed. We tried it with a factory reset radio and a new blank model with the same result. 

### Additional context
N/A

stronnag, 2023-08-15:
Please provide a log fiile ..... (or a X9D+ Radio)

stronnag, 2023-08-15:
Please also provide details of the radio firmware ....

b14ckyy, 2023-08-15:
> Please also provide details of the radio firmware ....

added LUA, OTX data and Firmware with EEPROM Dump.

stronnag, 2023-08-15:
Thanks, look forward to seeing the OTX (csv) log.

b14ckyy, 2023-08-15:
Oh sorry misunderstood you. Here you go.
[iNAV-Wing-2023-08-13.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12344640/iNAV-Wing-2023-08-13.zip)

stronnag, 2023-08-15:
Thanks, that looks horribly sane at first glance.

b14ckyy, 2023-08-15:
Yep, that one works.

stronnag, 2023-08-15:
It works with ELRS on EdgeTX as well, so we'll go with it.

b14ckyy, 2023-08-15:
But its interesting that this only seemed to be a problem on the X9D. Worked with no problem on 2.2.1 on the TX16S.

closed 2023-08-15 by stronnag

b14ckyy, 2023-08-15:
Oh good point. I have EdgeTX on mine maybe that's why it worked there.

stronnag, 2023-08-15:
> Oh good point. I have EdgeTX on mine maybe that's why it worked there.

I expect that is indeed the case ... train wreck