iNavFlight/OpenTX-Telemetry-Widget #117: Script syntax error

nlexfpv, 2023-08-19:
### Describe your question

### To reproduce

1. Go to '....'
2. Click on '....'
3. Scroll down to '....'
4. Should it be doing this?

### Expected behavior

### Screenshots

### Additional context

stronnag, 2023-08-19:
Please provide a coherent description "Go to '....'" is hardly helpful.

nlexfpv, 2023-08-19:
hi sir im new in fpv world i would to know how to fixed this issue 
already update the lastest in my radio taranis x9dplus inav lua came out
bad argument #1 to 'getValue' (string expected, got nil)

stronnag, 2023-08-19:
If you can't provide a coherent description of the issue, I can't help. Sorry.

nlexfpv, 2023-08-19:
[image]

stronnag, 2023-08-19:
* Firmware and version
* Script version

nlexfpv, 2023-08-19:
edgeTx 2.8
script 2.2.1

stronnag, 2023-08-19:
OK, now you need to explain exactly what you are trying to do ...

nlexfpv, 2023-08-19:
once i on the radio lua work just fine but once plug the battery it came out syntax error bad argument nil value

nlexfpv, 2023-08-19:
yes same problem happen..and im using matek f405 wse fc

stronnag, 2023-08-19:
INAV version?

Can you please generate an EdgeTX telemetry log?

Ate you using the builtin FrSky TX or some other external module (like CRSF) ?

nlexfpv, 2023-08-19:
inav 6.1.1

extenal module happy model es24tx pro

sorry sir do u have video for to generate those log? this radio is my 1st bougt.. before this im using flysky i6x

stronnag, 2023-08-19:
OK, CRSF. 

Can you please try installing the script from #113, you can find the zip file in https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/113#issuecomment-1679038715

nlexfpv, 2023-08-19:
ok i need to fly again for the log right..

will get back to u once i fly tomorrow

appreaciate ur time thanks

stronnag, 2023-08-19:
Did the script from #113  fix the problem ?

nlexfpv, 2023-08-19:
same problem

stronnag, 2023-08-19:
> ok i need to fly again for the log right..

You can do it without flying. This is an EdgeTX function (not an INAV log). A CSV telemetry log in the TX `LOGS` directory.  There is a little information in [our documentation](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Configuration-Settings.md#setup). 

There are plenty of tutorials (e.g. https://oscarliang.com/log-telemetry/) and you tube videos (Google search for "edgetx telemetry log").

nlexfpv, 2023-08-19:
[Shark Elrs.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/12386095/Shark.Elrs.zip)

stronnag, 2023-08-19:
The logs are dated 2000-01-23, so I guess you haven't set the clock?

They also contain NO useful telemetry or indication of telemetry. 

* The radio health fields (`1RSS(dB) | 2RSS(dB) | RQly(%) | RSNR(dB) | ANT | RFMD | TPWR(mW) | TRSS(dB) | TQly(%) | TSNR(dB)` are all zero
* There are NO telemetry sensors in the log. 

So the script cannot work if you haven't set radio or telemetry up correctly.

closed 2023-08-19 by stronnag