iNavFlight/OpenTX-Telemetry-Widget #120: Errore lua script inav

Alfonsomartina, 2023-09-13:
Salve.. Dopo aver aggiornato edgetx 2.9 su radiomaster tx16s e dopo aver inserito il lua script di inav 2.2.2 sul display il simbolo dei gradi della bussola e del variometro vengono sostituiti da un sibolo che raffigura una chiocciola@. Inoltre accedendo al lua script dininav tramite il pulsante sys si genera un errore syntax. 
[image]
[image]

stronnag, 2023-09-13:
https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Tips-And-Common-Problems.md#reporting-issues-on-github

stronnag, 2023-09-13:
> Hi.. After updating edgetx 2.9 on radiomaster tx16s and after inserting the inav 2.2.2 lua script on the display the degrees symbol of the compass and the variometer are replaced by a symbol depicting a snail@. Furthermore, accessing the dininav lua script via the sys button generates a syntax error.

1. The script has not changed, it has always used value 64 (which is indeed `@`). Perhaps the OS has changed the character set. for colour radios. On B&W radios, `°` is displayed correctly.
2. Expected behaviour. This is a telemetry script, you cannot run it from the SYS menu.

closed 2023-09-13 by stronnag