iNavFlight/OpenTX-Telemetry-Widget #125: Horusx12s EDGE TX no Menu vers. 2.2.3

uwe-air, 2023-10-30:
Hi i have the problem that i cant get in to the menu by moving the stick.

I use the version widget version 2.2.3 and edgetx 2.9.0

i went to the screen with the widget and the RF module is OFF than i moved the left stick to the right corner -> Nothing 
i tried that than with all possible variations on the Left and Right Stick -> nothing

This :
https://luatelemetry.readthedocs.io/en/latest/Configuration-Settings/
Horus/Jumper T16
Must be unarmed to access the config menu when running as a widget
Use the following (mode 2) stick controls to navigate the config menu:
sample
.......is is not working somehow.

The widget it self works fine with the sensors movement etc
I hope you can help me out

packetpilot, 2024-01-26:
Does holding the "model" button not bring up the menu while the widget is running?

stronnag, 2024-01-26:
Using  the stick works correctly in the Companion, as documented. 
If the hardware does not work the same as Companion, please seek the assistance of the firmware developers or radio manufacturer.

closed 2024-01-26 by stronnag

uwe-air, 2024-05-29:
Hi
Model Mode must be 1, (if you use 2,3,4 you need to change just for the setup, than bring the former throtle stick to the center and move the Left on up right. 

Solved