iNavFlight/OpenTX-Telemetry-Widget #128: On "Radiomaster Pocket" , request way to "Cycle through views" (no EVT_ENTER_BREAK key)

captcha1, 2023-11-16:
### Is your feature request related to a problem? Please describe

The code uses EVT_ENTER_BREAK to "Cycle through views", but the "Radiomaster Pocket (running EdgeTX 2.10.0-rm-build)" doesn't have that key.

### Describe the solution you'd like

Some other way to "Cycle through views".  A different key or a long press use the sticks or remap the keys ...

### Describe alternatives you've considered

### Additional context

stronnag, 2023-11-16:
Not yet a supported radio.
This will be considered after ETX 2.10 in released.

captcha1, 2023-11-16:
Any ideas for a workaround ? Maybe I could use a different key ? Maybe copy SCRIPTS/TELEMETRY/iNav.lua to iNav_pilot.lua and iNav_radar.lua and hard code it ...?

Thanks !

BTW, awesome stuff !

stronnag, 2023-11-16:
You're welcome to play around for you own solution, create a PR even. However, _this maintainer_ is only prepared to officially support: 
* Hardware that I own
* Hardware that is in an official ETX build, including Companion support.
* For Companion supported devices, then devices where Companion behaves the same as the real hardware (reportedly, there are cases where is does not).

Trying to support hardware without access to that hardware is a waste of time, both for the maintainer and the user.

Once there is an official ETX Companion for the radio, I will take a look at it.

pfeerick, 2023-12-14:
@captcha1 What happens when you press and release the roller (ENTER)? As for me that *is* switching from screen to screen. I'm running a recent nightly firmware build, so it is *possible* it was fixed subsequent to the factory firmware being made, but I don't remember offhand anything that aimed to fix that as an issue.

closed 2024-07-23 by stronnag