iNavFlight/OpenTX-Telemetry-Widget #129: v2.2.3 Edge Degree - script syntax error index to nil value

madox6257, 2023-11-23:
Hello, i tried to install the Lua telemetry widget, but i cannot select the widget in the select telemetry screen (fullscreen enabled). 
Running the script returns a script syntax error: "?.0: attempt to index a nil value".

I'm using the TX Radiomaster TX16S with EdgeTX 2.8.2. My FC is the MATEKF405SE with INAV 6.1.0.
I have no other Lua scripts running. 

I tried running the script with the debug tool in EdgeTX companion, but i found no instructions how to run a script in the simulator.

stronnag, 2023-11-23:
It works for me.
You need to provide a better explanation, as the combination of:
*   "i cannot select the widget in the select telemetry screen"
* and "running the script returns a script syntax error: "?.0: attempt to index a nil value"."

Makes no sense, as the **only** way the script works is as a telemetry script.

madox6257, 2023-11-23:
I can execute the script, if i select it in SYS/SD CARD/SCRIPTS/TELEMETRY/iNav.lua.
I have an iNav widget folder on my SD, but it wont appear with the other widgets.

I already reflashed EdgeTX, deleted the whole SD card and copied the SD content packet for 2.8. over.

stronnag, 2023-11-23:
> I can execute the script, if i select it in SYS/SD CARD/SCRIPTS/TELEMETRY/iNav.lua.

The you will get an error, because you're not meant to run telemetry scripts like that. This is expected.

closed 2023-11-23 by stronnag

madox6257, 2023-11-23:
Ok. Thank you. 
But why can't I select the widget in the telemetry screen?
Should I open a different issue, because it seems like, that's a different issue?