iNavFlight/OpenTX-Telemetry-Widget #130: Upon flight reset, "Attempt to call a nil value" error thrown

dxs94, 2023-11-26:
TX16S // ETX 2.9.2 // widget 2.2.3 Edge Degree release // no other LUA script running

Widget works fine. When clicking jog wheel, select "Reset telemetry", then select "Reset flight" the widget crashes (all the time) with "Attempt to call a nil value". The crash happens only when "resetting flight", this does NOT happen when only resetting telemetry.

stronnag, 2023-11-26:
Does it / did it  happen in 2.8?

stronnag, 2023-11-26:
(a) Does not happen here
(b) This is new bug  / mis-feature in EdgeTX 2.92 (quelle suprise!!!) > In prior version, it did not present the model menu, it presented the widget menu.
(c) as I'm fed up with working around new breaking features in EdgeTX, please complain to them (quote this Issue if you like).
(d) Don't press that option ...

closed 2023-11-26 by stronnag

dxs94, 2023-11-26:
Can't say. Went from OTX 2.3.15 straight to ETX 2.9.2, and updated the widget at the same time.
Will try to reinstall but not many hopes...

dxs94, 2023-11-29:
Well I think I found the fix. For some reason upon reset the radio is trying to load this file "load_e.lua" which does not exist. I then made a copy of "load.lua" which I renamed "load_e.lua" and it worked.

I then looked at previous 2.2.0 release and the file "load_e.lua" was indeed present. So maybe not an edgetx problem after all ;)
Thanks for keeping this great widget going forward.

stronnag, 2023-11-30:
Indeed, it should not try to load that obsolete file ...

stronnag, 2023-11-30:
Fixed by 2.2.4. Most of the recent releases here are due to non-backwards compatible EdgeTX changes ... somewhat tiresome.