iNavFlight/OpenTX-Telemetry-Widget #134: Pilot view 'ground shading' extremely faint on Taranis X9D+?

csurf, 2024-03-13:
the 'ground shading' on the pilot view is extremely faint on my taranis. See pic.

I tried messing with the contrast, color, and backlight settings on my radio, but nothing helped. What else should I try, or is this a bug?
I'm running latest OpenTX firmware 2.3.15.

[image]

stronnag, 2024-03-13:
Looks OK in companion, so there's nothing I can do further.
[image]

csurf, 2024-03-13:
it might look ok in the sim, but not in real life unfortunately.

@stronnag is the sim the only way you have of testing display output for the taranis? Is there any setting in the LUA script that I can try changing? If you have a taranis, what are your settings for contrast/color/backlight?

stronnag, 2024-03-13:
I do not have a Taranis. I expect that it it were a problem beyond your hardware, it would have been reported previously, as I'm aware of a number of Taranis users.

csurf, 2024-03-13:
Well, my taranis LCD is in perfect working order, so I'm not sure what to tell you.

What setting within the inav lua script controls the contrast/darkness of the ground shading? I would like to try changing it, recompiling and testing it...

closed 2024-03-13 by csurf