iNavFlight/OpenTX-Telemetry-Widget #142: Reversed Roll in attitude indicator (artificial horizon) Graphic

packetkiller, 2024-04-15:
Describe the bug
The iNAV Telemetry Widget is displaying the attitude indicator graphic roll in reverse (left roll is right roll and right roll is left roll).

TX = RadioMaster TX16S
TX Firmware = 6.17
EdgeTX ver = 2.9.3 "Providence"
RX = TBS Crossfire Diversity Nano
RX Firmware = 6.17
Telemetry Protocol = Crossfire
Flight Controller = Lumenier LUX-H7 HD Flight Controller
FC Firmware = iNAV 7.0.1 with inav_7.1.0_MATEKH743.hex

To reproduce
Turn on all electronics (TX, VRX, Battery connected to aircraft)
While looking at the aircraft from the rear (the aircraft pointing away from you in the direction it would normally fly), lower the left wing (simulate a left roll).
Look at the iNAV Telemetry Widget on the TX and notice the roll graphic depicts a right roll.
Repeat for right roll.
Again, look at iNAV Telemetry Widget on the TX and notice the roll graphic depicts a left roll.
Note, all telemetry data indicates good values with the exception of the reversed roll attitude indicator graphic.
Expected behavior
An aircraft left roll is expected to show a left roll in the iNAV Telemetry Widget. Same for right roll.

Screenshots
See attached files for left and right roll with iNAV Telemetry Widget showing reverse roll in attitude indicator graphic.

Radio and model settings
The model name is Opticopter1.
Note: the other model using the same TX (an ARPro1) now also displays incorrectly in the iNAV Widget since iNAV upgrade to 7.1.0.
Attached is the .dat and EdgeTX .etx settings backup files and other log files.

Additional context
Also provided is a copy of the iNAV config settings for Opticopter1.
[TX_LOGS_Opticopter2-2024-04-15-134032.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/14984464/TX_LOGS_Opticopter2-2024-04-15-134032.zip)
[FC_LOGS.zip](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/files/14984467/FC_LOGS.zip)

stronnag, 2024-04-15:
Duplicate of 131

closed 2024-04-15 by stronnag

packetkiller, 2024-04-15:
Not a duplicate. New model. Same issue.
Please do not just dismiss this without at least providing some evidence there is NOT an issue. Call it education on my part if you want.

sensei-hacker, 2024-04-15:
If you want use to take out time to help you understand, please stop wasting our time by creating duplicates all over the place.

MrD-RC, 2024-04-16:
If the plane banks left. The AHI will bank right. This is so that the ground looks level if you were in the aircraft. 

If this is the way yours is working. It is correct. If not. Post a video showing the LUA screen and your aircraft. Then bank your aircraft.