iNavFlight/OpenTX-Telemetry-Widget #145: Unable to enter configuration settings on Radiomaster TX12 v2

AlexCrosby, 2024-05-01:
EdgeTX version: 2.8.0
Transmitter model: Radiomaster TX12 v2
LUA version: 2.2.4

Hi
This seems to be similar to the issue reported here:
https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/17

Except that this is with the v2 version of the radio.

On long press of the roller button you get the expected Reset telemetry/Reset flight menu, but hitting back does not display the config menu, it just goes back to the lua widget.

**To reproduce**

1.     Press the TELE button to start the widget
2.     Try long pressing roller to enter configuration opens the Reset telemetry/Reset flight menu
3.     Pressing back goes back to the widget

stronnag, 2024-05-01:
That's not what the [current documentation] (https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Configuration-Settings.md#other-radios)  states.

Perhaps you could try that.

AlexCrosby, 2024-05-01:
> That's not what the [current documentation] (https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Configuration-Settings.md#other-radios) states.
> 
> Perhaps you could try that.

Thanks. I have assumed https://luatelemetry.readthedocs.io/en/latest/Configuration-Settings/ was up to date, which only mentions the roller wheel. Thanks

closed 2024-05-01 by AlexCrosby