iNavFlight/OpenTX-Telemetry-Widget #146: Inverted Pitch display on artificial horizon (roll axis is ok)

akex03, 2024-06-04:
### Describe the bug

On the artificial horizon display, the pitch axis is inverted. 
I'm using a Jumper T16 (edgetx-firmware-v2.9.4), crossfire protocol with Matek ELRS-R24-D,  and flight controller lefei sparrow 2.

### To reproduce

When the FC is moved pitch-up, the artificial horizon displays a pitch-down animation. When the FC is moved pitch-down, the artificial horizon displays a pitch-up animation.

When I go to my telemetry page on my controller, I see that the ptch telemetry is postive when the flight controller is in pitch-up and negative when the flight controller is in pitch-down. @

### Expected behavior

The artificial horizon animation should match the FC movement/position

### Radio and model settings

MrD-RC, 2024-06-04:
Can you please describe what you see on the AHI. When you tilt the nose of the aircraft up. The ground of the AHI should get lower in the display.

akex03, 2024-06-04:
Hi Darren,When I tilt the nose of the aircraft up, the ground portion of the AHI is getting bigger (blue portion getting smaller) and when I tilt the nose of the aircraft down, the ground portion of the AHI is getting smaller(blue portion getting bigger).Hope this helps,Thanks,

MrD-RC, 2024-06-04:
If you check in INAV Configurator's setup page. Does the on screen model react the same as the real aircraft?

akex03, 2024-06-04:
Hi Darren,How can I access the Inav configurator's setup page? Is it the one that you place the right stick centered and the left stick up/right? Because in that menu there is only settings.Thanks,

Mysteur-Yo, 2024-06-07:
Hi Akex03

you need to connect you model (FC board) to the computer with an USB cable 
then with INav configurator software you can all the parameter of your model and also the movement of the model

the reaction that you describe is look like the FC is mount in your model in reverse position 

[image]

akex03, 2024-06-11:
Hi,

I'm using a Leifei Sparrow 2 flight controller connected to a crsf/elrs telemetry receiver, the flight controller outputs the following telemetry:  GPS, GSpd, Hdg, Alt, Sats, Rxbt, Curr, Capa, Bat%, Ptch (inverted), Roll, Yaw, FM, 1RSS, 2RSS, RQLY, RSNR, ANT, RFMD, TPWR, TRSS, TQly, TSNR

There is no FC mounting positions configurations. There is only one way to install it.
The telemetry output for pitch is as follows:

  *
When the aircraft nose goes up (pitch-up) the pitch telemetry is positive (units in RAD)
  *
When the aircraft nose goes down (pitch-down) the pitch telemetry is negative (units in RAD)

Is this the normal way?

Thanks,

Alexandre

image.png (view on web)<https://github.com/iNavFlight/OpenTX-Telemetry-Widget/assets/172062678/9e0e515e-48dc-49f6-a488-29b7f1a218df>

MrD-RC, 2024-06-11:
The Leifei Sparrow 2 isn't running INAV. This LUA Script is for INAV flight controllers only. We have no control over what data other systems provide.

closed 2024-06-11 by MrD-RC

akex03, 2024-06-12:
Ok perfect thanks all for your support,Have a nice day,Alexandre

ChenSihao-cee, 2025-06-16:
My iNav has the same issue: the data shows correctly in the ground station, and the pitch reading is normal there. But the pitch angle output in the telemetry is reversed compared to what's displayed on the ground station. Any idea how to fix this, mate? Love from China.