iNavFlight/OpenTX-Telemetry-Widget #147: Color Issue with the widget under Edge-TX

Mysteur-Yo, 2024-06-07:
### Describe the bug
the colors of the background and text are the same, that means I can't see the information's of the telemetry 

### Screenshots
with the default colors system of Edge-TX :
[image]

with custom colors system :
[image]

If I shut down the model I can see the text in red due to the telemetry lost, that means the text is present in the script 
[image]

### Radio and model settings
My radio is the Jumper T18 runing wint Edge-TX V2.9.4
[image]

Thanks.

stronnag, 2024-06-07:
And your version of the widget?

Mysteur-Yo, 2024-06-07:
Your question push me to download again and over-right the file widget and script 
now that work very well without color issue

thanks this topic can be close or delete

stronnag, 2024-06-07:
Indeed, you were using an obsolete, 2 year old version.

stronnag, 2024-06-07:
The version is shown at satrtup

closed 2024-06-07 by stronnag