iNavFlight/OpenTX-Telemetry-Widget #148: Error in refresh(): bad argument #1 to 'getValue' (string expected, got nil

Yodabutt2, 2024-06-10:
### Describe your question
Radiomaster TX 16S, EdgeTX 2.10.1 (upgraded from 2.8.0 in troubleshooting this issue), LUAC installed, iNav LUA 2.2.4

### To reproduce
I get this error on my screen's iNav Widget immediately after the receiver connects.
I do see telemetry data on the radio in the model settings.
I have copied the LUA files multiple times and confirmed the LUAC option is installed.
I haven't been able to find any information online about this error message.
No other LUA's should be running. No other widgets on this model.
There is plenty of space on the SD card.

Thanks for any suggestions.

stronnag, 2024-06-11:
Can you please provide the following:
* INAV CLI `diff`
* EdgeTX CSV log

Yodabutt2, 2024-06-13:
My apologies for the delay. 
Here is the iNAV Diff and I will the EdgeTX log this weekend.
[SD-INAV_7.1.2_cli_20240609_102822.txt](https://github.com/user-attachments/files/15829871/SD-INAV_7.1.2_cli_20240609_102822.txt)

Thank you.

Yodabutt2, 2024-06-14:
Hey,  Here are EdgeTX logs. I started the logging, connected the battery to the model, and stopped logging a couple of seconds after getting the LUA error. It always occurs as soon as I get my Telemetry connected message on the radio.

[Nano Goblin-2000-01-01-000619.csv](https://github.com/user-attachments/files/15842734/Nano.Goblin-2000-01-01-000619.csv)
[Nano Goblin-2000-01-01-000729.csv](https://github.com/user-attachments/files/15842735/Nano.Goblin-2000-01-01-000729.csv)
[Nano Goblin-2000-01-01-000759.csv](https://github.com/user-attachments/files/15842736/Nano.Goblin-2000-01-01-000759.csv)

Let me know if I can provide anything else.
Thank you.

stronnag, 2024-06-14:
And the specific RX/TX you're using please.

Yodabutt2, 2024-06-14:
HappyModel ES24TX Pro TX module and RadioMaster RP2 RX.
Thanks.

stronnag, 2024-06-14:
At first glance, looks like a duplicate of #117.
The header is missing at least one required field (`Sats`)

stronnag, 2024-06-14:
### Working example of ELRS telemetry Ids:
```
Date Time 1RSS(dB) 2RSS(dB) RQly(%) RSNR(dB) ANT RFMD TPWR(mW) TRSS(dB) TQly(%) TSNR(dB) Ptch(rad) Roll(rad) Yaw(rad) FM VSpd(m/s) GPS GSpd(m/s) Hdg(°) Alt(m) Sats RxBt(V) Curr(A) Capa(mAh) Bat%(%) Rud Ele Thr Ail S1 6P S2 EX1 EX2 EX3 EX4 LS RS SA SB SC SD SE SF SG SH SI SJ LSW CH1(us) CH2(us) CH3(us) CH4(us) CH5(us) CH6(us) CH7(us) CH8(us) CH9(us) CH10(us) CH11(us) CH12(us) CH13(us) CH14(us) CH15(us) CH16(us) CH17(us) CH18(us) CH19(us) CH20(us) CH21(us) CH22(us) CH23(us) CH24(us) CH25(us) CH26(us) CH27(us) CH28(us) CH29(us) CH30(us) CH31(us) CH32(us) TxBat(V)
```

### Your telemetry Ids:
```
# 619
 Date Time RxBt(V) Ptch(rad) FM VSpd(m/s) 1RSS(dB) GPS Rud Ele Thr Ail P1 P2 P3 SL1 SL2 SA SB SC SD SE SF SG SH LSW CH1(us) CH2(us) CH3(us) CH4(us) CH5(us) CH6(us) CH7(us) CH8(us) CH9(us) CH10(us) CH11(us) CH12(us) CH13(us) CH14(us) CH15(us) CH16(us) CH17(us) CH18(us) CH19(us) CH20(us) CH21(us) CH22(us) CH23(us) CH24(us) CH25(us) CH26(us) CH27(us) CH28(us) CH29(us) CH30(us) CH31(us) CH32(us) TxBat(V)
# 729
 Date Time RxBt(V) Ptch(rad) FM VSpd(m/s) 1RSS(dB) GPS Rud Ele Thr Ail P1 P2 P3 SL1 SL2 SA SB SC SD SE SF SG SH LSW CH1(us) CH2(us) CH3(us) CH4(us) CH5(us) CH6(us) CH7(us) CH8(us) CH9(us) CH10(us) CH11(us) CH12(us) CH13(us) CH14(us) CH15(us) CH16(us) CH17(us) CH18(us) CH19(us) CH20(us) CH21(us) CH22(us) CH23(us) CH24(us) CH25(us) CH26(us) CH27(us) CH28(us) CH29(us) CH30(us) CH31(us) CH32(us) TxBat(V)
# 759
 Date Time RxBt(V) Ptch(rad) FM VSpd(m/s) 1RSS(dB) GPS Rud Ele Thr Ail P1 P2 P3 SL1 SL2 SA SB SC SD SE SF SG SH LSW CH1(us) CH2(us) CH3(us) CH4(us) CH5(us) CH6(us) CH7(us) CH8(us) CH9(us) CH10(us) CH11(us) CH12(us) CH13(us) CH14(us) CH15(us) CH16(us) CH17(us) CH18(us) CH19(us) CH20(us) CH21(us) CH22(us) CH23(us) CH24(us) CH25(us) CH26(us) CH27(us) CH28(us) CH29(us) CH30(us) CH31(us) CH32(us) TxBat(V)
```
If you're not able to provide coherent telemetry, the widget cannot work.
This may be caused by your ELRS settings not providing enough capacity for telemetry; it is not something the widget can fix.

closed 2024-06-14 by stronnag