iNavFlight/OpenTX-Telemetry-Widget #149: No Text when connected to RX

jlt206, 2024-06-23:
Edge-tx16s JB - vers. 2.7.1 - INAV v.7.1.1 - CRSF

When the RX is connected, I loose all the information text and it changes the theme?

[image]
[image]

stronnag, 2024-06-23:
What version of the widget, looks like an ancient, unsupported version.

stronnag, 2024-06-25:
Please feel free to reopen if you have anything useful to add.

closed 2024-06-25 by stronnag

onurboyuk07, 2024-07-08:
i'm having the same problem myself.is there a solution

stronnag, 2024-07-08:
> i'm having the same problem myself.is there a solution

Not without a modicum of useful information, in particular the version of the widget you're using.

onurboyuk07, 2024-07-09:
1.7.2
1.7.5

stronnag, 2024-07-09:
Unsupported, obsolete, unofficial and known broken versions. 

You should consider upgrading .

jlt206, 2024-07-09:
So the answer is that this is the wrong version to use and the correct version can be found in the INAV Git Repositry here: https://github.com/iNavFlight/OpenTX-Telemetry-Widget

Like a lot of people I searched for the widget and it bought me here. Seems logical, but here it is no longer supported or updated as far as I can tell.