iNavFlight/OpenTX-Telemetry-Widget #154: Jumper T15 - Emergency Mode when selecting iNav widget in telemetry setup

dingrgh, 2024-07-20:
### Description

Jumper T15 (480x320 colour touch screen)
EdgeTX 2.10.2
iNav Lua Telemetry 2.2.4

When setting up the telemetry widget on a new model, selecting the iNav widget immediately causes the TX to crash and go into Emergency Mode.

### To reproduce
On TX, install LuaTelemetry v2.2.4.
In EdgeTX, create a new model.
Select Telemetry and add a new screen (full screen with no tabs or trim displays). Then attempt to select iNav as the widget to be used.
Boom! Emergency mode.
After powering down and up, you are back to a new model with no telemetry screens (other than the default blank one).

stronnag, 2024-07-20:
Supported radio  ... didn't think so.

stronnag, 2024-08-05:
Can you please try latest master and let me know if it fixes the T15 problem? (#155)

CCNNCHUN, 2024-08-06:
I'm using jumper t15 firmware version edgetx2.10.3
lua 2.2.4 and it still doesn't work
[image]

stronnag, 2024-08-06:
I asked you to test master, not 2.2.4

CCNNCHUN, 2024-08-06:
> ~我让你测试master，不是2.2.4~
> 
> 抱歉，请测试师傅。它应该由 #155 修复
I tested the master and now it works fine, thanks

CCNNCHUN, 2024-08-06:
> ~I asked you to test master, not 2.2.4~
> 
> Apologies, please test master. It should be fixed by #155

I tested the master and now it works fine, thanks

stronnag, 2024-08-06:
Thank you. New release on its way ...

closed 2024-08-06 by stronnag