iNavFlight/OpenTX-Telemetry-Widget #156: About Flysky EL18 screen display issues

UGun11, 2024-08-07:
### Describe the bug
The FLYSKY EL18 script screen cannot be displayed in all portrait orientations.
[image]

### To reproduce
[image]

### Expected behavior

### Screenshots

### Radio and model settings

### Additional context

stronnag, 2024-08-07:
Not a supported device

AmaZz0r, 2024-09-07:
same problem here.
was hoping it would work as I thought it would be somewhat the same as the nv14

AmaZz0r, 2024-09-07:
@UGun11 I think I found a (better than current state) solution.
I edited the initialization script (iNav.lua), to make the script think that it is running on a NV14.
Meaning that it seems to mostly display proper portrait mode now.
Before pasting/transferring the new files, I deleted all of the old files belonging to the widget, so basically a "clean install".
As you can see below, I attached an image of how it looks on my EL18 at the moment and an archive containing the edited file.

[image]
[LuaTelemetry_v2.2.5_lua.zip](https://github.com/user-attachments/files/16919129/LuaTelemetry_v2.2.5_lua.zip)

closed 2025-01-04 by stronnag