iNavFlight/OpenTX-Telemetry-Widget #161: I can not open configuraion setting in radiomaster tx16s

zhangtongxie, 2025-01-04:
My radio is radiomaster tx16s ,and I used INVA lua in my radio ,I can use it great ,but I do not open configuration setting by anyway

1.Make lua Full Screen
2.Press sys or mdl or long press the roller or open it by rocker
3.I only see the sats

open configuration setting succesfully
[image]

stronnag, 2025-01-04:
https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Configuration-Settings.md

closed 2025-01-04 by stronnag

zhangtongxie, 2025-01-05:
> https://github.com/iNavFlight/OpenTX-Telemetry-Widget/blob/master/docs/Configuration-Settings.md

I find it in INAVButtons but ,when I try to use it I can not open .such as
[image]

Uploading VID_20250105_192143.mp4…
[image]