iNavFlight/OpenTX-Telemetry-Widget #169: edgeTX 2.11 uses Lua 5.3

arkarkark, 2025-04-04:
### Describe the bug

edgeTX 2.11 currently on [release candidate 3](https://github.com/EdgeTX/edgetx/releases/tag/v2.11.0-rc3) uses lua 5.3 (not 5.2 as previously) so the precompiled `.luac` files shipped in `LuaTelemetry_v2.3.0.zip` no longer work. 

### To reproduce
1. Update radio to edgeTX 2.11 RC3 or later (I did mine on a Radiomaster GX12)
2. unzip `LuaTelemetry_v2.3.0.zip` onto the sdcard
3. try and add iNav to the telemetry screens (mdl, page <, screen x, script, iNav)
4. See error when you try and go to that page after hitting `tele`

### Screenshots

[image]

arkarkark, 2025-04-04:
I tried to build my own, but alas I get this error

[image]

to build it myself. I had to install lua 5.3 which was a little annoying.

```
brew tap homebrew/core --force
brew edit lua@5.3
# comment out the line that says disable in it
HOMEBREW_NO_INSTALL_FROM_API=1 brew install  lua@5.3
```

then I updated the Makefile like this

```
LUA_DIST = /opt/homebrew/opt/lua@5.3
LUA = $(LUA_DIST)/bin/lua
LUAC = $(LUA)c
```

it all built fine and the files in the zip archive identify as lua 5.3 but it doesn't work on my radio. looks like it's making l4888 and I want l4444. I'll work out how to do that. I don't want to upload each one to [this tool](https://lua-bytecode.github.io/)

stronnag, 2025-04-05:
Just install the lua version (which will be compiled on the device,, as instructed on the [Release Page](https://github.com/iNavFlight/OpenTX-Telemetry-Widget/releases/tag/v.2.3.0).

arkarkark, 2025-04-05:
of course! sorry. thanks.

closed 2025-04-05 by arkarkark