iNavFlight/OpenTX-Telemetry-Widget #172: sytax error on connection.

dhr7v, 2025-05-01:
A getValue method is struggling with a nil value. Running the telem screen and script works fine until the drone gets connected upon which it throws the syntax error.
Tried all the warnings for previous releases. I'm on a radiomaster pocket. Sorry if I missed something

stronnag, 2025-05-01:
* Works fine on my Boxer
* No useful information provided

closed 2025-05-01 by stronnag

dhr7v, 2025-05-08:
I think u mentioned sats, I dont have any satellite data im sure coming in the telem since theres no gps, could an exception handler perhaps be incorporated
    Radiomaster pocket
    EdgeTX RM factory (1fdb58ba)
   All drones
No other widgets, 32gb sandisk sd card

https://github.com/iNavFlight/OpenTX-Telemetry-Widget/issues/148 referenced above