iNavFlight/OpenTX-Telemetry-Widget #186: OLED remote control horizon display reversal

boqiansteam, 2025-09-25:
The monochrome OLED remote control's default pixel display is the inverse of an LCD screen, which causes the script's HUD horizon to appear inverted. It is hoped that this issue can be resolved by adding a color inversion setting or by providing specific adaptation for the OLED remote control. Thank you very much.(σ≧︎▽︎≦︎)σ。

MrD-RC, 2025-09-25:
Are you sure? There have been many instances of people reporting issues with the AHI. When they are just reading it incorrectly. If the aircraft rolls right (left wing high) \. The left side of the AHI should go down and the right side should go up /

boqiansteam, 2025-09-25:
[image]
Just like this, the black part of the horizon is on top, and the white part is on the bottom. I believe it is incorrect.